motioneye-project/motioneyeos #2602: Pi Zero W Securing/Hiding Wi-Fi Password

ivi-lab, 2020-12-08:
I'm using my pi zero w in an outdoor environment and whilst unlikely as its screwed in place, the thought of someone pinching it crossed my mind. In theory if someone takes my SD card its not difficult to gain access to my network as the password is visible in plain text.

So can the wifi password be protected somehow?

I've thought about some workarounds but they're not ideal- like creating a separate wifi network.

On a side note, a huge thanks to the dev for this software!

ivi-lab, 2020-12-08:
I thought I'd browse my sd card on my pc and now I feel a bit silly posting this. I assume the wpa_supplicant.conf file gets eaten after the first setup as its no longer visible.

closed 2020-12-08 by ivi-lab

starbasessd, 2020-12-08:
Not quite true, the wpa_supplicant.conf files is moved to
/etc/wpa_supplicant/wpa_supplicant.conf.
I use(d) the program
wpa_passphrase [your_passphrase] [enter]
which will generate a hash for your use:
psk=[generated text] which you can use instead of a password. Doesn't do
much good, as any computer using wpa_supplicant.conf can use it when they
find it.
If someone were to hijack the PiZero, I'd immediately change it. I'd use
ACL for the MAC Address on it's own separate wifi network access point.
Lots of better ways to secure it, that would give you a head start to make
security changes in the event of theft.
Example:
https://unix.stackexchange.com/questions/40/use-wpa-supplicant-without-plain-text-passwords

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?