motioneye-project/motioneyeos #2603: Action Buttons on MotionEyeOS not working (An error occurred. Refreshing is recommended) #1964

zapp-bit, 2020-12-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026.

### Board Model

I am using the following board/model: Raspberry PI 3B+.

### Camera

I am using the following type of camera: V4L2

My camera model is: Chinese eBay special, logitech webcam

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

*12v alarm (has its own power supply, activated by the pi triggering a relay)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5672730/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5672731/motioneye.log)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

I have attached all the required files. 

Hi there.

As the title states, the action buttons do not work, when pressed, I get the error "An error occurred. Refreshing is recommended"

I can confirm the script works, when executing it from command line, it works a charm. The issue seems to be with how the web gui is trying to execute it. I have attached the relevant log information below.

2020-12-10 18:18:51: [motioneye] ERROR: [Errno 8] Exec format error Traceback (most recent call last): File "usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception File "usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 1685, in post File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 1688, in run_command_bg File "usr/lib/python2.7/subprocess.py", line 394, in __init__ File "usr/lib/python2.7/subprocess.py", line 1047, in _execute_child OSError: [Errno 8] Exec format error 2020-12-10 18:18:51: [motioneye] ERROR: 500 POST /action/1/alarm_on/?_=1607584731568&_username=admin&_signature=e13c31ce11861bcdaf5dfabc5fcf05023fa2ab80 (192.168.1.50) 20.30ms

The script is a copy/paste of your basic script to turn a light on and off, and I am using a raspberry pi 3 b+
copied from here - https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons

GPIO=18
test -e /sys/class/gpio/gpio$GPIO || echo $GPIO > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio$GPIO/direction
echo 1 > /sys/class/gpio/gpio$GPIO/value

Please tell me if you know how to fix this.
Thanks

starbasessd, 2020-12-10:
OSError: [Errno 8] Exec format error
2020-12-06 11:54:06: [motioneye]    ERROR: 500 POST /action/1/alarm_on/?
To fix it, just add the shebang e.g., if it is a shell script; prepend #!/bin/sh at the top of your script:

This is why I said post the actual script you are using.

starbasessd, 2020-12-10:
In your case, you are probably missing the 

#!/bin/bash

at the beginning of one of the scripts.

starbasessd, 2020-12-10:
https://stackoverflow.com/questions/27606653/oserror-errno-8-exec-format-error

zapp-bit, 2020-12-11:
Well now I feel a bit silly. I am so sorry.... Rookie Mistake.
This has resolved my issue. 

Thanks for the quick response

closed 2020-12-11 by zapp-bit

starbasessd, 2020-12-11:
No problem, glad to help.