motioneye-project/motioneyeos #2604: problème d'identifant

titof2375, 2020-12-12:
hello, I have a camera where I have identification problems, my identifiers are good for both with the last version
[image]

starbasessd, 2020-12-12:
It would be very helpful for you to answer the questions when you opened
the issue.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2020-12-12:
WHAT?

starbasessd, 2020-12-12:
Which version of motionEyeOS? 20190911, 20200203, 20200606, dev20201026?
Which Raspberry Pi? Zero, 1, 2, 3, 3B+ 4B 1GB 2GB 4GB 8GB?
What manufacturer and model Camera?
How is the Pi connected to the network?
How is the camera connected?

titof2375, 2020-12-12:
ha ok the version
motionEye Version 0.42.1
Motion Version 4.3.1
OS Version motionEyeOS dev20201026
with a 4 giga rpi
the rpi and the camera in rj45
 and the type of camera
[image]

starbasessd, 2020-12-12:
I would switch back to 20200606 then, if that was working correctly,
otherwise I would make sure the username & password for the camera are in
the correct case.
For example username is Admin vs admin vs AdMiN
Password is password vs Password vs PAssWoRD

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2020-12-13:
hello even with the version you told me it does not work, on my web page no problem, even with zoneminder and my home automation software no problem.

starbasessd, 2020-12-13:
Take it back to the version that worked. You did not state which version that was. If it was 20190911 then use that one.

Your screen shot of the camera doesn't say anything of what manufacturer or model.

Please describe your setup, fully.
Running what on what. Running which devices.

For example:
I run several separate networks: My primary Gigabit ethernet, my primary 11ac wifi, my motionEye 11b/g/n wifi and ethernet, and a couple virtual networks.
I have 5 primary cameras: 
3 D-Link DCS-5020L PTZ cameras on their own Wifi network.
1 FosCam 99xx PTZ camera on ethernet
1 TIGA Generic Dashcam connected to USB on Computer
1 PiZeroW with PiCam on D-Link Wifi network for testing purposes (motionEyeOS dev20201026 currently, but 3 others able to be switched in)
1 i5 Lenovo SFF PC as primary Hub (Production) Running Ubuntu 20.04, and motion/motionEye installed as of 2020-11-15, with motionEye 0.42.1, motion 4.3.1
1 Pi4B-4GB on ethernet as test hub running motionEyeOS dev20201026, pulling feeds from the 4 wireless cameras directly, and the USB cam via stream from hub.
1 Pi3B+ on ethernet as test hub running motionEyeOS 20200606, pulling feeds from the 4 wireless cameras directly, and the USB cam via stream from hub.
I have a VirtualBox server running at times several different versions of Linux with various installs of motion/motionEye and can build a new VM quickly for testing issues. it has a Ryzen 7, 32GB RAM, 10 TB storage, 4 gigabit ethernet connections, and 5-8 VMs running various services.
I have a 6 TB Buffalo NAS as an FTP/SMB test bed and production backup storage.
I have a VM on my Desktop PC to run Pi-Gen (the system Raspberry Pi Foundation Foundation uses to create their images of Raspberry Pi OS.
I try to reproduce new issues when people present them here, if it has not been discussed before.

I don't expect people to go into that much detail when describing their setup, but the more info you can give, the better we can help you.

titof2375, 2020-12-13:
I can give you access to my house if you want

starbasessd, 2020-12-13:
Nope. Nein. Nichts, Negative, Non. No way.Het.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2020-12-13:
OK no worries

titof2375, 2020-12-13:
I just dismantled my camera and I found that it is a focam model fi8904w

starbasessd, 2020-12-13:
Choose network Camera
URL will be:
*rtsp://[ip_address]:150/videoMain*
*use your username and password*

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2020-12-13:
[image]

titof2375, 2020-12-13:
[image]

starbasessd, 2020-12-13:
My FosCam uses
rtsp://ip_address:88/videoMain
Heres a good website to look for your setup:
https://www.ispyconnect.com/man.aspx?n=Foscam
If you click on a section, it will pop up a url generator which you can use
to copy/paste into VLC
If you copy paste to here (take out username and password or chage it to
admin & password, I should be able to help you to edit for motionEye

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

titof2375, 2020-12-13:
good I found, I made about thirty tests with the site https://www.ispyconnect.com/, and it is almost that at the end that I found the right ur
l **http: / /192.168.1.245/videostream.cgi?user="user"&pwd="pwd".&resolution=32**

closed 2020-12-13 by titof2375