motioneye-project/motioneyeos #2605: CSI 8MP Camera Module not working

ZeroDot1, 2020-12-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20201026

### Board Model

I am using the following board/model: Raspberry PI 3

### Camera

8MP SonyCam, https://www.ebay.de/itm/8MP-Camera-Module-3280x2464-160-IMX219-Chip-Camera-Board-Support-Night-Viewing/283867413888?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649 

Type: CSI Camera Module
Chip: IMX219 (for Sony) 1/4
Size: Approx. 25 x 24 x 17mm / 1 x 0.9 x 0.7in
Resolution: 3280 x 2464
Field of View: 160°
Can support night viewing

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

no

### Log Files
[image]

starbasessd, 2020-12-13:
Was it auto-discovered on boot?
Can you attach the boot.log, dmesg log, motioneye log?
How was the camera added if not auto-discovered?
How does it show in Video Device? What for Camera Device and Camera Type?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ZeroDot1, 2020-12-13:
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5684691/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5684693/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5684694/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5684695/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5684696/motion.log)
[image]

starbasessd, 2020-12-13:
dmesg log says it sees an attached USB camera, is it working correctly?
motioneye log showing lots of failures to connect to (192.168.178.34)
Did the system discover your CSI camera as an MMAL device?
If you can, use the console or SSH into the system, and run the command
logrotate /etc/logrotate.conf [enter]
This will clear the logs.
Disregard/delete all other cameras you are trying to set up.
delete the CSI camera if it is not Camera 1.
Disconnect any USB cameras.
Only have the CSI camera connected.
unplug power, and then plug power back in.
Does the system discover the CSI camera?
Does the system try to create any cameras, and what Camera type (V4L2 or
MMAL)?
If it wasn't auto-discovered, try adding a MMAL type camera.
Then send the dmesg, and motioneye logs again.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

rivsc, 2020-12-21:
Your camera is detected ? 

Try 
`sudo vcgencmd get_camera`
you should see
`supported=1 detected=1`

Stop motioneye `service motioneye stop` and try to get picture from raspistill (no error, good news !) : 
`raspistill -o test.jpg`
Restart motioneye : `service motioneye start`

Try to reduce resolution in motioneye (640x480).
Try to use "Local MMAL camera".

starbasessd, 2020-12-21:
be sure to select Local MMAL camera if manually adding.

ZeroDot1, 2020-12-23:
Thanks for your answers, I'll look at it again as soon as possible.

ZeroDot1, 2020-12-29:
OK, I have removed all cameras except the camera module
Here are the logs and screenshots.
I have reseted the logs and then restarted.
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5751084/motioneye.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5751087/dmesg.log)

[image]
[image]

starbasessd, 2020-12-29:
Also run the command
vcgencmd get_camera [enter]
either in the console or ssh session.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ZeroDot1, 2020-12-29:
vcgencmd get_camera says: supported=1 detected=1

[image]

rivsc, 2020-12-29:
Can you take picture with raspistill ?

`raspistill -o test.jpg`

(I have the same problem with exactly same result. My official camera module is, after 2 years outside, not completely broken but not really functional...).

ZeroDot1, 2020-12-29:
`# raspistill -o test.jpg`

```bash
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

```

starbasessd, 2020-12-29:
Can you test with an install of RaspberryPiOS?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ZeroDot1, 2020-12-29:
Not at the moment, sorry.

ZeroDot1, 2020-12-29:
I think the camera should work because it's new, but I'm sure it's a bug or missing driver that's causing the problem.

starbasessd, 2020-12-29:
It is quite possible the camera is defective, even though it shows with
vcgencmd get_camera
Just a diagnostic.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ZeroDot1, 2020-12-29:
If possible I will test this with RaspiOS as soon as possible.

ZeroDot1, 2020-12-30:
[image]
[image]

starbasessd, 2020-12-30:
Probably defective camera board. It does see it, but can't use it. The
PiCamera V2 is very nice, available many places, and is inexpensive ($25/US)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ZeroDot1, 2021-02-18:
This problem is solved, I close it here.

Thank you.

closed 2021-02-18 by ZeroDot1

starbasessd, 2021-02-18:
What was the issue/fix?

gh-ls, 2021-06-18:
I'm having the same issue with the same camera on Pi3 and Pi4, with OS Version motionEyeOS 20200606.
Can I know the solution? Thanks.

starbasessd, 2021-06-18:
I would try dev20201026 for both the Pi3 & Pi4. 
I have a feeling it wasn't supported by PiOS, either, and he got a different camera.
Did you run 
vcgencmd get_camera
and what was the result?

gh-ls, 2021-06-18:
Hi, thank you for the very prompt reply! Is it possible to run the command from inside MotionEyeOs?

starbasessd, 2021-06-18:
No, only from Console (keyboard & mouse) or SSH session.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

gh-ls, 2021-06-18:
Ok, I'll try the dev20201026 also. Now my wife's calling for dinner... I'll tell you the results, thank you.

gh-ls, 2021-06-18:
Using  dev20201026 on Rpi4, now.  vcgencmd get_camera : supported=1 detected=1

starbasessd, 2021-06-18:
It should have auto-detected the camera and set it up with basic settings.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jhall0326, 2022-05-16:
I have similar issue.  Rpi4 Motioneyeos dev20201026 imx219.  Camera works fine on a Bullseye build.  vcgencmd get_camera reports supported=1 detected=1.  MotioneEyeOS detects it and sets it up as MMAL Camera with Camera Device=vc.ril.camera but within the MotionEyeOS GUI it is a gray square saying "Unable to open video device.

Motion.log:
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
[1:ml1:Camera1] [ERR] [VID] create_camera_component: camera component couldn't be enabled
[1:ml1:Camera1] [ERR] [VID] vid_start: MMAL cam failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)

starbasessd, 2022-05-16:
@jhall0326 Out of curiosity, what resolution & frame rate are you trying to run it at?
I don't refer to the supported max resolution and max frame rate. When testing 'new to me' cameras, I usually test a 5 frames per second, and a max of 1280x720 (16x9) or 1024x768 (4x3) to confirm operation. Then I increase until I get those grey screens. I also use H.264 (non-OMX), or MPEG4 (.avi) for testing.