motioneye-project/motioneyeos #2606: Unable to mount a USB HDD on my router for File Storage

DLeikis, 2020-12-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-12-15:
No info?

DLeikis, 2020-12-15:
I'm trying to, but I can't seem to edit the issue to add to it.

starbasessd, 2020-12-15:
So reply to it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-15:
Network Share (first storage option) is not supported. 
For Network Share, it must be a Windows SMB or CIFS share.
If you wish to use FTP (port 21) you need to turn on 
Upload Media Files (just below where you have picked Network Share).

starbasessd, 2020-12-15:
Without knowing if your router supports SMB / CIFS / Windows shares because you didn't mention make & model of the router it'll be hard to help you set that up.

DLeikis, 2020-12-15:
Do I disable the Storage device?
What is the location entry under Upload Media Files?

DLeikis, 2020-12-15:
My router is a TP-Link Archer A7 v5.0.  I am unable to see any selections or menu options for SMB.  I have no issues with accessing with any function or sharing this drive under Win10.

starbasessd, 2020-12-15:
No, do not disable the Storage Device, infact you need to set it back to default:
Storage Device  Custom Path
Root Directory   /data/output/Camera1

Turn On Upload Media Files
Upload Service FTP Server
Server Address 192.168.0.1
Server Port  21
Location   PicsShare
Include Subfolders  ON
Username  (whatever username can access the share)
Password (for the username above)
Click Apply at the top BEFORE clicking Test Service, then click Test Service

It needs the Storage device ON Defaults to create the picture and/or videos before it can upload to the FTP server on your router.

starbasessd, 2020-12-15:
" I have no issues with accessing with any function or sharing this drive under Win10. " because you are forcing FTP by using the address:21

starbasessd, 2020-12-15:
https://www.tp-link.com/us/support/faq/1527/  (case 1) setup SMB Shares
https://www.tp-link.com/us/support/faq/2385/ (case 2) if your settings don't look like the ones in Case 1
Well written, step by step instructions. Things have to be set up for it though
If you get it set up, let me know, and I'll walk through the setup for Network Share.

DLeikis, 2020-12-15:
After making the FTP settings and Applying them, I test the service, but receive the error "550 PicsShare - Permission Denied" .

I have carefully verified the UserName and Password, but same effect.  

I will read the articles for SMB Shares.

DLeikis, 2020-12-16:
I have read and completed the correct faq article, but I still do not have access via FTP on MotionEyeOS.  I continue to receive the error stated above "550 PicsShare - Permission Denied".

starbasessd, 2020-12-16:
What are your entries under File Storage, Upload Media Files?
you can just put 'username' and 'password', I don't want you to post those
here...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DLeikis, 2020-12-17:
[image]

I hope this screen scrape works...
Otherwise, I will just type it.

DLeikis, 2020-12-17:
The entry /Test is while debugging.  Usually it is the volume name of the drive.

starbasessd, 2020-12-17:
the location used for Storage Device is invalid Because of that you
probably aren't generating any images to save to ftp.
Default is Custom Path and /data/output/Camera1
Location for FTP should be /Test. (ftp uses / for directory separators)
Folder name is case sensitive, too. (is the share folder name Test?
If, in a web browser, you type
ftp://192.168.0.1 [enter]
What shows up? Is Test an option?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DLeikis, 2020-12-17:
[image]
[image]
Here are the present settings in Motion and the router settings.  Still getting Permission Denied error.

DLeikis, 2020-12-17:
If I do as you ask ----  
If, in a web browser, you type
ftp://192.168.0.1 [enter]
What shows up? Is Test an option?

Nothing shows up in the browser window.  I can do anything under windows when the drive is mapped  to \\PicsShare.

starbasessd, 2020-12-17:
My network NAS is at 192.168.2.252
If I type ftp://192.168.2.252
in Edge or FireFox, I get prompted for my username and password, then I get
both sharepoints
I can then access anything under them
If I do the same thing with one of my motionEyeOS Pi's addresses
Again I get prompted for username & password (admin and [blank])
and I get 2 sharepoints  sdcard and storage

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DLeikis, 2020-12-17:
That does not happen for me.  I will try to find another way to test it.

starbasessd, 2020-12-17:
I would recommend FileZilla, but you have to jump though a couple of hoops,
not worth it for testing IMO
Chrome browser forces Edge (IE) to be opened to work.
I just created a random ftp server on a Ubuntu 20.04 VM, and it uploaded
there, no problem.
I'll leave it here, and maybe someone else can give some ideas.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DLeikis, 2020-12-17:
Thank you for the help.

DLeikis, 2020-12-17:
An update on my HDD access problem.

I was successful at full access to the drive and storing both snapshots and movies to the proper folders after changing the storage location to the Windows device name.  I had called that drive "Photos" when mapping a Win10 drive to that device and it wasn't until I used FileZilla and saw that this was the name it used that I made the realization.  It was not looking for the ftp name, rather the Windows pathname.

Again, thank you for your help and suggestions!  Regards!

starbasessd, 2020-12-17:
Good to hear it's working, don't forget to close the issue.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-12-17 by DLeikis