motioneye-project/motioneyeos #2607: Fresh installation on Raspberry Pi Zero W: "starting motioneye: failed"

dissidente, 2020-12-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: **motioneyeos-raspberrypi-20200606**.img.xz.

### Board Model

I am using the following board/model: **Raspberry Pi Zero W v1.1**.

### Camera

I am using the following type of camera: Not quite sure (check [this link](https://www.aliexpress.com/item/4000080232431.html?spm=a2g0s.9042311.0.0.27424c4dVqn3XV) for the hardware).

My camera model is: see previous information.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

No peripherals

### Log Files

I consider the following log files relevant to this issue:
 
[boot.log](https://github.com/ccrisan/motioneyeos/files/5698189/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5698202/dmesg.log)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/5698203/messages.txt)

### Issue description

I'm trying to install motionEye to my RPi0W, and seem to have an issue while starting motionEye (literally).

From the top:

* Tested the 32GB SD card (W/R) - OK
* Burned the image to SD card
* Configured and added wpa_supplicant.log file with wireless information for my network
* Booted the board connected to my TV for feedback
* 1st run seems to go ok, and install & configure the required stuff
* ... but after 1st run all boot ups after fail with the message "starting motioneye: failed" and then reboot, into a loop

It connects successfully to the wireless network (my dhcp server has a record for it), although I haven't tried to connect to the board by any network protocols.

I've tried (for troubleshooting sake) to disconnect the camera flat ribbon, but the behavior is the same.

Any suggestions/ideas/solutions/beer ?

Thanks in advance

starbasessd, 2020-12-15:
Weirdness. Can you post your wpa_supplicant.conf file here (cut out your
password) please? (hopefully you just typo'd the name of the file above)
Bootlog shows it never connects to anything.
What other changes did you make?
Try dev20201026 for PiZeroW
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-15:
Was the WebGUI working?
How did you get the log files?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dissidente, 2020-12-16:
Hi @starbasessd 

Indeed. But I can confirm that although it fails starting motioneye, it has wireless connectivity since it configures ap0 to use dhcp (it gets a dhcp ip lease and is pingable).

Here is the screen shot of the boot sequence:

[image]

Did not do any change, besides dropping the wpa_supplicant.conf file, which is show to be ok by the connectivity test.

Could not connect to the WebGUI dough. I managed to access the log files by connecting the card to my NAS to read the fs.

I've tried to burn the dev image and this one works. It boots ok and starts motioneye, and I can access the WebGUI (didn't try to login, but hey, it proves a point =) )

Will I have an upgrade path from the dev image to the next stable one, or should I consider this image a test step and try to further troubleshoot the latest stable image?

Thanks for your help  =)

starbasessd, 2020-12-16:
I have my PiZeroW with piCam up on 20200606.
mywpa_supplicant.conf consists of:
-----
[root@meye-c0e188a9 etc]# cat wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="ssid"
    psk="password"
        key_mgmt=WPA-PSK
}
-----
ip a shows:
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast qlen
1000
    link/ether b8:27:eb:b4:dd:fc brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.120/24 brd 192.168.2.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:feb4:ddfc/64 scope link
       valid_lft forever preferred_lft forever
3: ap0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop qlen 1000
    link/ether b8:27:eb:b4:dd:fc brd ff:ff:ff:ff:ff:ff
I don't know what's not working...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dissidente, 2020-12-16:
Oops, you're right. I forgot to share my wap_supplicant.conf:

```
country=PT
update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
            scan_ssid=1
            ssid="ssid"
            psk="password"
}
```

But, again, this shouldn't be a networking issue (at least not directly). Even when motioneye fails to start the board has an IP address assigned and reachable.

I looked for any additional relevant logs where I could find details on motioneye startup but couldn't find any (at least in the mounted fs of the card, while connected to my NAS).

starbasessd, 2020-12-16:
Try changing the wpa_supplicant.conf to:

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=PT

network={
    ssid="ssid"
    psk="password"
    key_mgmt=WPA-PSK
}

You only need to use scan_ssid in an AP rich environment.
The biggest 'normal' cause for failure to start is not getting an IP
address on a PiZeroW

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dissidente, 2020-12-18:
Oddly enough, you're right. Changing my wap_supplicant.conf to yours did the trick.

I find it odd because clearly before starting motioneye the system is already connected to wifi, as it gets a dhcp lease, gets time, and responds to pings.

Thanks for your help @stuartsummers  =)

closed 2020-12-18 by dissidente

starbasessd, 2020-12-18:
<sniffle> got my name wrong, but good to hear you are up and running.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

timmeh87, 2021-06-29:
Hey guys I had exactly this problem and It was because I had pulled the power plug during the initial setup, which takes a rather long time to complete.  Why did I do that? Because I forgot to put my supplicant file on the drive... Re-flashing motion eye os and letting the initial setup complete fixed my problem. So if anyone is  coming here with troubles with a fresh image and you think the power might have been turned off within 5 minutes of initially turning on, then try reflashing from scratch and waiting longer for initial setup

JohnWiik, 2023-01-12:
@starbasessd hi there, I added os.conf and wpa_supplicant.conf with code as 

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=PT

network={
    ssid="ssid"
    psk="password"
    key_mgmt=WPA-PSK
}

However, the first booting showed "Starting motioneye: done" but then immediately started to reboot. During the rebooting, the screen only shows "Starting motioneye: _ ", then reboots again and again showing the same thing. I wonder if you can help me with this problem. Thank you!

P.S. I added os.conf file is because it was showing “Configuring wired network: no device
Panic: rebooting in 100 seconds (caused by network) …….”. 

[image]
[image]

coder0107git, 2023-01-12:
@JohnWiik Did you add the following to both your `os.conf` and `wpa_supplicant.conf`?
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=PT

network={
            ssid="ssid"
            psk="password"
            key_mgmt=WPA-PSK
}
```
It should **only** be in your `wpa_supplicant.conf` file. Please see the [wiki](https://github.com/motioneye-project/motioneyeos/wiki/) for the [`os.conf`](https://github.com/ccrisan/thingos/wiki/os.conf#:~:text=These%20files%20contain%20settings%20that%20affect%20the%20functioning%20of%20the%20entire%20operating%20system.%20os%2Dplatform.conf%2C%20if%20present%2C%20should%20contain%20platform%2Dspecific%20OS%20settings%20(as%20opposed%20to%20common%20OS%20settings).) and [`wpa_supplicant.conf`](https://github.com/ccrisan/thingos/wiki/wpa_supplicant.conf#:~:text=The%20file%20is%20not%20present%20by%20default%20and%20should%20be%20created%20by%20the%20user%2C%20if%20a%20WiFi%20connection%20is%20needed.) file documentation.

JohnWiik, 2023-01-12:
No, it was only added to wpa_supplicant.conf. The code in os.conf is just OS_NETWORKLESS="true".

When I boot the Zero W the first time, it worked fine. However, I couldn't find the IP address so I shut down the system by unplugging the power. But after that, I couldn't make the booting work anymore. Do you think maybe something corrupted my SD card due to the unplugging? Thank you!

coder0107git, 2023-01-12:
It's possible that the SD Card became corrupted when it was unplugged. When I was beginning with Pi's, I corrupted several OS installs and had to reflash the SD Card because I didn't shut the system down beforehand. I think your problem is having `OS_NETWORKLESS="true"` in your `os.conf`. The line is explained [here](https://github.com/motioneye-project/motioneyeos/wiki/WiFi-Access-Point#:~:text=motionEyeOS%20still%20needs%20a%20network%20connection%20to%20function%20well%3B%20if%20you%20really%20don%27t%20want/need%20one%2C%20set%20os_networkless%20to%20true) (TL;DR `OS_NETWORKLESS="true"` means that wifi and ethernet will be disabled for normal network connectivity). If you do eventually get it to boot and can't find the ip address then [this youtube video clip](https://www.youtube-nocookie.com/embed/ybQzlvukSBo?start=280&end=293&autoplay=true) might help.