motioneye-project/motioneyeos #2609: Motioneyeos won't boot without ethernet connection

dayiethan, 2020-12-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: V4L2 (not totally sure about this but it is just a Raspberry Pi v1.3 Camera Module plugged directly into the Pi.

My camera model is: 5MP 1080P Video Camera Module

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi and Ethernet (I have the wpa_supplicant file and I don't want to use ethernet but the pi won't boot without ethernet but will function normally without ethernet after booting)

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

Not log files but the terminal has these lines when booting with ethernet:

Starting wpa_supplicant: done (brcmfmac)
Configuring wireless network: 10.0.0.254/24
Configuring wired network: dhecp

The first two lines still appear when booting without ethernet but then the configuring wired network fails and it panic reboots.

starbasessd, 2020-12-20:
Please add boot.log, motioneye.log, messages.log
Make sure your wpa_supplicant.conf file follows the format:
-----
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="ssid"
    psk="password"
key_mgmt=WPA-PSK
}
-----
Change country to your country (and must be uppercase)
make sure your ssid only contains letters, numbers, and '-'

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-20:
BTW, your camera should be auto-discovered or MMAL

dayiethan, 2020-12-20:
> Please add boot.log, motioneye.log, messages.log
> Make sure your wpa_supplicant.conf file follows the format:
> -----
> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
> update_config=1
> country=US
> 
> network={
>     ssid="ssid"
>     psk="password"
> key_mgmt=WPA-PSK
> }
> -----
> Change country to your country (and must be uppercase)
> make sure your ssid only contains letters, numbers, and '-'
> 
> 
> 
> 
> > Preliminary Docs
> >
> > I confirm that I have read the CONTRIBUTING
> > guide before opening this issue.
> >
> > I confirm that I have read the FAQ
> > <https://github.com/ccrisan/motioneyeos/wiki/FAQ> before opening this
> > issue.
> > motionEyeOS Version
> >
> > I am running motionEyeOS version: 20200606
> > Board Model
> >
> > I am using the following board/model: Raspberry PI 3B+
> > Camera
> >
> > I am using the following type of camera: V4L2 (not totally sure about this
> > but it is just a Raspberry Pi v1.3 Camera Module plugged directly into the
> > Pi.
> >
> > My camera model is: 5MP 1080P Video Camera Module
> > Network Connection
> >
> > My motionEyeOS unit is connected to the network via: Wifi and Ethernet (I
> > have the wpa_supplicant file and I don't want to use ethernet but the pi
> > won't boot without ethernet but will function normally without ethernet
> > after booting)
> > Peripherals
> >
> > I am using the following peripherals that I consider relevant to this
> > issue:
> >
> > None
> > Log Files
> >
> > I consider the following log files relevant to this issue:
> >
> > Not log files but the terminal has these lines when booting with ethernet:
> >
> > Starting wpa_supplicant: done (brcmfmac)
> > Configuring wireless network: 10.0.0.254/24
> > Configuring wired network: dhecp
> >
> > The first two lines still appear when booting without ethernet but then
> > the configuring wired network fails and it panic reboots.
> >
> >
> 
> 
> Thanks
> 
> Kevin Shumaker
> 
> 
> Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>
> 
> N38° 19' 56.52"
> W85° 45' 8.56"
> 
> Semper Gumby
> “Don't tell people how to do things. Tell them what to do and let them
> surprise you with their results.” - G.S. Patton, Gen. USA
> Ethics are what we do when no one else is looking.
> Quis custodiet ipsos custodes?
> “There is no end to the good you can do if you don’t care who
> gets the credit.” - C Powell
> You know we're sitting on four million pounds of fuel, one nuclear weapon
> and a thing that has 270,000 moving parts built by the lowest bidder. Makes
> you feel good, doesn't it?

My apologies, I'm pretty new to all this so I don't quite know how to get those logs but this is an image of the boot screen:
[image]

Also, my camera didn't show up automatically but I was able to add it. Not sure if that's a problem the camera works fine now and streams (the images I take however don't upload to the Google drive I set up unless connected to Ethernet).

dayiethan, 2020-12-20:
My wpa_supplicant should also be correct, I downloaded a template and replaced everything appropriately.

starbasessd, 2020-12-20:
Did you try to force a static IP address?
Most of the templates around aren't updated to the recent Buster format.
Please try mine.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-20:
If you have the Pi up on ethernet, you can go to the WebGUI and under
Expert Settings, there are several log file links there that you can click
to download.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dayiethan, 2020-12-20:
I did force a static IP address and I can access it. I have attached the motioneye.log and boot.log now. Thank you!
[boot.log](https://github.com/ccrisan/motioneyeos/files/5721447/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5721448/motioneye.log)

starbasessd, 2020-12-20:
Why does your static have an IP of 10.0.0.52/24 but your gateway has a
192.168.1.1 address? That is why it's panicking. It can't see the
192.168.1.1 router.
If your wifi router is issuing IP addresses, then you need to point the
static_ip static_gateway to the wifi router IP address.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dayiethan, 2020-12-20:
Ah I see, but when I first booted with the wpa_supplicant and used Angry IP scanner to find suitable IP addresses the 10.0.0.something was what was on my wireless network. For reference my wireless network that I configured my wpa_supplicant to is the same network that my ethernet is plugged into.

starbasessd, 2020-12-20:
So your primary network (connecting to your modem/router is 10.0.0.x?
Then your static_gateway needs to be your modem/router ip address (usually
10.0.0.1 or 10.0.0.254)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dayiethan, 2020-12-20:
Yeah, my static IP is 10.0.0.254 currently

starbasessd, 2020-12-20:
Your Router address?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dayiethan, 2020-12-20:
Sorry, I meant the static IP for my Raspberry Pi wireless connection

dayiethan, 2020-12-20:
And Angry IP Scanner automatically scans for the range 10.0.0.0 to 10.0.0.255 on my wifi network. I see multiple other devices on there as well

starbasessd, 2020-12-20:
The static_ip.conf file, in your case should be:

STATIC_IP="10.0.0.254/24"
STATIC_GW="10.0.0.1"
STATIC_DNS="8.8.8.8"

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dayiethan, 2020-12-20:
Hmm how can I do that? I didn't know that that was an issue that I had to address? I know that I am supposed to be able to boot motioneyeos without ethernet connection. Will this fix that? Thank you by the way!

starbasessd, 2020-12-20:
IF you use a valid static_ip.conf and a valid wpa_supplicant.conf file,
then yes, it will boot to wifi with a static IP address on your wifi
without the need for an ethernet connection.
I do it a lot of times in testing for other users.
I personally use my router to issue the static IP to most of my Pis via the
DHCP request from the Pi, that way I can take them out of the home network
and still have them work.
Most routers can issue the same IP address every time they see a DHCP
request from a specific MAC address.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dayiethan, 2020-12-20:
Gotcha. I really only plan on using this one my home wifi so that really shouldn't be a huge problem. How do I use a valid static_ip.conf? I really don't know what or where that is. Sorry!

starbasessd, 2020-12-20:
https://github.com/ccrisan/thingos/wiki/static_ip.conf
goes into the /boot partition right after you image the sdcard along with
the wpa_supplicant.conf file

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dayiethan, 2020-12-20:
I know that the wpa_supplicant file doesn't work if I've already booted motioneyeos before. Will I need to reflash for this one to work as well? Is there a template for this file as well because I don't really understand that page you linked. Also can I just copy it paste it directly into the sd card folder like the wpa_supplicant file?

starbasessd, 2020-12-20:
Yes.
I've attached the file I pasted in before.
Yes, copy/paste the same as the wpa_supplicant.conf file

dayiethan, 2020-12-20:
Ahhh thank you so much! I got it working now and the google drive back up also works over wifi as well!

closed 2020-12-20 by dayiethan