motioneye-project/motioneyeos #2610: Checking Pi Core Temp and resetting Pi user password

dayiethan, 2020-12-21:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: local V4L2

My camera model is: Raspberry Pi NOIR Camera Module V2

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

I am worried that my Pi may be overheating a bit so I wanted to check the core temperature by SSHing into the pi via Putty. First of all, is this still a viable way to check even with motionEyeOS flashed on? Also, when SSHing it kept denying my password for the pi user (I tried raspberry and most of the possible passwords that I would have changed it to). Am I doing something wrong or is there a way to reset the pi user password with motionEyeOS flashed on?

Thank you!

starbasessd, 2020-12-21:
User is admin, and if you didn't set a password in the WebGUI, there is no
password.
Also, the messages log (under Expert Settings, has the temp logged now.
If you are concerned when you ssh in, be sure to get vcgencmd
get_throttled, also, it'll warn you if it overheated or under-volted since
the last reboot.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dayiethan, 2020-12-21:
Ah, so when I have motionEyeOS flashed on, in order to ssh I have to use the admin log-in rather than the pi user log-in? Also, vcgencmd is what I was going to try and use once I was SSHed into the pi, are you saying that there is a better way to do that? To add a bit more context, I am merely plugging in the NOIR camera and letting it run and the pi is getting a bit hot (not sure if it's overheating per say but it is getting pretty warm) but the camera isn't getting hot. Is this a problem?

starbasessd, 2020-12-22:
There isn't a pi user in motionEyeOS.
Yes, you must use the admin user to log in, just as in the WebGUI
If you want to see it without ssh'ing in, you can check the messages log.
If you do ssh in, be sure to run the vcgen get_throttled command along with
the vcgencmd measure_temp command to see if it throttled itself due to heat.
If it is anything but 0x0 as the result for get_throttled, you can get an
idea what the trigger was by going to this site:
https://harlemsquirrel.github.io/shell/2019/01/05/monitoring-raspberry-pi-power-and-thermal-issues.html

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

dayiethan, 2020-12-22:
Great! Thank you so much!

closed 2020-12-22 by dayiethan