motioneye-project/motioneyeos #2611: Network Share not connecting

Mike1236, 2020-12-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 

OS Version | motionEyeOS 
Raspberry Pi4 hub dev20200907
Raspberry Pi Zero W 20200606

### Board Model

I am using the following board/model: 
Raspberry Pi4 for hub
Raspberry Pi Zero W

### Camera

I am using the following type of camera: Simple MJPEG Camera on the Raspberry Pi Zero W

My camera model is: Raspberry Pi Camera

### Network Connection

My motionEyeOS unit is connected to the network via: Hub is ethernet, camera is wireless

### Log Files

I consider the following log files relevant to this issue:

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5737429/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5737430/boot.log)

I have a Windows desktop running 24/7 that I use as a NAS. I connected a 1TB external hard drive to it. Currently it's named Camera F:\ with a folder inside called Recordings. I tried pointing the Network Share on the Pi hub to the IP of the Windows desktop which is 192.168.1.17 and Share Name as "Recordings".

I have tried these following combinations for Network Server:
\\Winserver
\Winserver
Winserver\
\\Winserver\f
\\192.168.1.17
\192.168.1.17
192.168.1.17

Along with the Share Name as:
Camera
Recordings
F:\
F

I have have tried all the SMB protocol versions. I created a new user on the Windows desktop called "Pi" with a password. SMB protocol is enabled in Windows features. I have no idea why it's not being connected. Much appreciated if anyone can help.

 
[image]
[image]

starbasessd, 2020-12-23:
For the Root Folder it should be /Recordings/, For the Server it should be
192.168.1.17
Unless you jumped through a bunch of hoops to disable SMB 3.0 or 3.1.1 in
Windows, you need to set your SMB version to 3.0 or 3.1.1 (depends to what
your current upgrade level is in Windows)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-12-23:
So I just tried

Server: 192.168.1.17
Root Folder: /Recordings/
SMB on 3.0 and 3.1.1
Both didn't work. I didn't touch any SMB 3.0 or 3.1.1 settings, I only enabled SMB 1.0.

Mike1236, 2020-12-23:
I want to add that I am able to see \\192.168.1.17\f from another Windows computer

starbasessd, 2020-12-23:
By default, Windows 10 uses SMB 3.1.1. To connect to it, you need to switch
SMB to 3.1.1 on the Pi

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-23:
Wait a second. It's saying you don't have any space available on /data. Can
you use the console or SSH into the Hub where you're trying to connect from?
If you can, run the command df -h, and see how much space is available in
the line:
/dev/mmcblk0p3            6.1G     32.7M      5.7G   1% /data (this is ona
a 8GB sd card in a Pi3B+ with motionEyeOS dev20201026)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-12-23:
Correct the SD card is full because I accidently set to store recordings to there.

Yes I have set SMB to 3.1.1 on the Pi hub. Still no go. :( 

[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/5737504/motioneye.1.log)
[image]

starbasessd, 2020-12-24:
Yeah, clean some space before it'll work.
You can search and destroy from the console/ssh if you need to.
Might be easier to start over. Re-image.
Connect your share before doing anything else, other than adding the camera.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-12-24:
Alright, I'm just going to reimage it since its much easier that way and I have nothing to lose right now, it's not fully setup to my likings yet. I'll update soon.

Mike1236, 2020-12-24:
Just reimaged. Looks like we're progressing. Now I'm getting this error:
Accessing network share failed: cannot create root directory!

Network Server: 192.168.1.17
SMB: 3.1.1
Share Name: Recordings
User: Pi
Password: ....
Root Directory: /Recordings/

fyi, I do have a folder called "Recordings" inside the drive, but that is also being shared. Not sure if this causes problems?

Mike1236, 2020-12-24:
Also, I'm curious how does it know to point to the F drive? I currently have a D and F drive, and I want it to store the recordings to the F drive specifically.

starbasessd, 2020-12-24:
When you created the share, did you point the share to F:?

So,
Created new share in Windows
   Created folder in device
   Right click folder, Properties
   Sharing tab
   Advanced Sharing
   Enable Share this folder
   Share Name = Recordings
   Permissions
   Add
   Add user pi
   Click Full Control
   Apply at the bottom
   Apply at the bottom
   Apply at the bottom
In the Pi:
Settings, File Storage:
Storage Device = Network Share
Network Server = 192.168.1.17
SMB = 3.1.1
Share Name = Recordings
Share Username = pi (case sensitive to whatever you created in Win10)
Share Password = whatever (also case sensitive)
Root Directory = /
Apply << MUST do first before Test
Test
I just did these steps on my Win 10 box share
Working.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mike1236, 2020-12-24:
You're a genius dude. Looks like I named it incorrectly in the beginning, the shared name was called \\Winserver\f. That was probably why it wasn't working. But now I see it as \\Winserver\Recordings. I'm going to try and see if it actually records some video now.

Mike1236, 2020-12-24:
Thanks for the help again. I can confirm it's now working. So I was curious. What happens if my Windows desktop needs to restart for an update. Does motioneye just stop recording for the timeframe that the storage share is off? Then it automatically just starts recording again when the share comes back up?

starbasessd, 2020-12-24:
Theoretically, yes. I have seen issues where the system is uploading, and
causes issues, but if it is not actively uploading, it shouldn't notice. If
the Share is down when it needs to upload, you'll run into this issue again.
If it becomes an issue, you could run a cron job to upload and delete the
local copy on a preset schedule. You wouldn't do it during the times you
apply and reboot the Windows machine.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-12-25 by starbasessd