motioneye-project/motioneyeos #2612: On boot - Configuring wired network: no link

alpew, 2020-12-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (Same issue with both 20200606 and 20200203).

### Board Model

I am using Raspberry Pi 4 - 4GB Ram

### Camera

IP cameras connected using RTSP

### Network Connection

My motionEyeOS unit is connected to the network via: wired Ethernet IPv4 using static IP 192.168.1.1/24

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

### Log Files

Unknown

**Issue description:** 
Load the motioneyeos for RPI4. Initial boot after install is good and I can connect to the motioneyeos from a computer browser across the network using the ip address of the RPI4. This proves the wired Ethernet port is working as expected connecting to the LAN using a Ethernet switch.

I configure a IP network camera using an RTSP string and the camera connects as expected. 
On rebooting the RPI running motioneyeos boot loads past rainbow screen as expected and starts booting the motionEyeOS until it gets to:

Configuring wired network: <response> = 'no link'
Boot continues presenting login prompt. See screenshot. 
[image]
Wireless is not configured.

**Further analysis**
Logging in and running ifconfig: see screenshot.

The above issue happens with both release versions mentioned above.

I am running motioneyeos on a RPI 3b and it runs perfectly. I wish to migrate my setup to the RPI4 but hit this wired network not being recognised after the initial boot and a camera has been added [block.](url)

Any advice gratefully received. I am not a programmer so forgive me my lack of programming knowledge.
 
[image]

starbasessd, 2020-12-24:
> Configuring wired network: = 'no link'

It is picking up an address, but only the ipv6 address. Can your
router/dhcp also issue a ipv4 address, or set a static IPV4 address? I
believe the program is looking for the ipv4 address to indicate the
connection is good, rightly or wrongly.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

alpew, 2020-12-24:
Yes that is my assessment too. No idea why it overrides the fixed IPv4 address configured in the MotionEyeOS gui. 

The RPI is configured with a static IPv4 address 192.168.1.220. On initial boot, it boots up as expected allowing me to access motioneyeos (MEOS) via a web browser from another computer on the LAN.  This is reflected in the Network section of MEOS gui interface. 

I don’t change the network setting but setting up a camera and rebooting then seems to lose the fixed ipv4 address as per the screenshot. Is there a way of setting a ipv4 address from the command Line or run the MEOS gui directly from the command line so I can access the Network section?

I do have a DHCP Server on my LAN which can issue ip addresses but I set the RPI4 with a fixed address for use with MEOS.

Al

starbasessd, 2020-12-24:
A couple of thoughts:
Upgrade to dev20201026 and see if it happens there.
Have your DHCP server issue the static IP you want. My router does it so I
can use DHCP on my devices and know where they are at home, but if I go
elsewhere, they automatically pull an address for that location.
You can set your static ip on first boot using a static_ip.conf file like
you would force turn on ssh and use a wpa_supplicant.conf file, by adding
it into your /boot partition after a clean iimage.
The priority is WiFi (so don't apply a wpa_supplicant.conf file in /boot)
create a static_ip.conf file in /boot:
-----
STATIC_IP="192.168.1.220/24"
STATIC_GW="192.168.1.1"
STATIC_DNS="192.168.1.1"
-----
Change STATIC_GW to your primary router
Change STATIC_DNS to whatever the rest of your network uses
Use a Linux-friendly editor like Notepad++, don't use Notepad if doing it
on a Win10 machine.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-24:
BTW, another advantage of having the DHCP server issue the statics is if
you need to make changes, only need to do it on the DHCP server, instead of
each individual device. And another is if you have to replace the DHCP
server, they'll all grab valid IP addresses from the new one. So, if your
current network is 192.168.1.x but the new DHCP server is defaulted to
192.168.42.x, at least the devices will come up until you can make all the
other changes needed...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

alpew, 2020-12-26:
Thanks very much for the ideas @starbasessd I will give it a go and report back. 

I use a Dreytek router and that allows me to bind mac addresses to certain IP addresses, effectively issuing  DHCP clients I set up in the binding table with the same IP address.

If anything it highlights there is a bug in the current releases. Possibly going back to the last 2019 release (or first 2020 can't remember) where some IPv6 changes were made I think. I haven't been able to load a release previous to those changes as my Pi Imager (Balenaetcher) keeps saying the file I download is corrupt. Possibly a problem with my unarchiver I use but not sure yet. Need to do some more research.

alpew, 2020-12-28:
Hi @starbasessd 

I've set up my RPI4 MAC address in the binding table on my router so that it automatically receives the correct ip address on boot. This has left the MEOS unchanged and set to 'Automatic (DHCP).

I've not yet created any conf files so everything else is as per the image (20201026)

I've done some testing this morning and I have isolated what causes the network link failure to occur.

In 'Expert Settings' there is an on/off option called 'Network Link Watch'. On my RPI3 I have set this to off. Setting this to off on the RPI4 version causes the 'Configured wired network' to fail on boot up with the 'no link' message.

I am going to configure the MEOS now with cameras and leave the Network Link Watch untouched at 'On' to see if all is ok.

starbasessd, 2020-12-28:
Interesting. I will look at it later and try to confirm.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

alpew, 2020-12-29:
Yes all cameras configured and working as expected. So the Network Link Watch option seems to be the cause of my no link issue.

PsyEng, 2020-12-31:
Is it possible that you've set [networkless](https://github.com/ccrisan/thingos/wiki/os.conf#os_networkless) to true? Maybe its activating itself by setting "Network Link Watch" to off.

alpew, 2021-01-01:
@PsyEng @starbasessd 

Happy New Year.

I've run this morning's test on releases 20201026, 20200606 and 20190911 (the first release with RPI4 support) and get identical results everytime.

I've loaded a fresh unchanged copy of MEOS, which boots as expected set as DHCP. 

I login to the GUI and make one change. Setting the 'Network Link Watch' to Off. Then I reboot.
For all 3 releases the result is the same as previously reported, with 'Configured Wired Network' returning 'No Link' on boot.

I've copied a screenshot showing the boot processes and the os.conf file for 0606 and 1026 releases.
Both show that the OS_NETWORKLESS setting in os.conf = 'false' @PsyEng 
[image]
[image]

Hope the above helps. It seems the RPI4 and the RPI3 are different in how they interpret the Network Link Watch setting.

PsyEng, 2021-01-01:
@alpew I wish you a happy new year, too. 

> Both show that the OS_NETWORKLESS setting in os.conf = 'false' 

It was just an idea. Could you upload your boot and dmesg log files? Boot with the option 'Network Link Watch' enabled or disabled, copy the logs to another place (bootlog likes to get overwritten) and then upload both here, maybe there's a hint. 

> It is picking up an address, but only the ipv6 address. Can your router/dhcp also issue a ipv4 address, or set a static IPV4 address? I believe the program is looking for the ipv4 address to indicate the connection is good, rightly or wrongly.

It's just a link-local address (fe80:, like 169.254.), so this doesn't matter.

alpew, 2021-01-01:
@PsyEng 

Where are the logs held so I can find them from the command prompt? The issue is I can't connect to the MEOS by browser as that obviously relies on the network link, and hence can't change the Network Link Watch setting from off to on without reimaging the SD card and starting again from scratch.. Unless you can do it from the command prompt?

The MEOS is set to DHCP IPv4 and I have not changed that setting. Wireless is also off as per the default config.

starbasessd, 2021-01-01:
Before doing anything with the logs, it's handy to run the following
command, to clear them:
logrotate /etc/logrotate.conf [enter]
Then start the reboot/settings process, that way you don't have a bunch of
really old log info.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PsyEng, 2021-01-01:
> Before doing anything with the logs, it's handy to run the following command, to clear them: logrotate /etc/logrotate.conf [enter] Then start the reboot/settings process, that way you don't have a bunch of really old log info.

Good point, didn't think of that. 

> Where are the logs held so I can find them from the command prompt?

They should be under /var/log/
You can copy them to /home/ftp/storage/
And, it's not a command prompt, it's a terminal 🙂

alpew, 2021-01-04:
Hi @PsyEng @starbasessd 

Thanks for the info and my apologies ref Terminal. :-)

You'll have to forgive me as I am not an expert on The Linux/MEOS commands.

So I have done the instructions ref the log files and copied them to /home/ftp/storage/ as instructed. So this is where I get stuck. I assume when I image the MEOS onto the memory card that it will overwrite the files on it including /home/ftp/storage. So how do I get the files off the Pi and onto a USB drive to copy across onto my MAC so I can send to you before I image the drive?

I can't access it from my network (as that is the original problem) and I've tried trying to find an extra memory card using fdisk I've plugged into a USB port. I can't find the drive as fidsk returns scrolls of lines but I can't work out how to paginate them so I can try and look at them in more detail.

FYI The boot.log is as per the screenshot I provided before. The dmesg.log is very long and that probably will be more use to you.

starbasessd, 2021-01-04:
https://www.maketecheasier.com/mount-access-ext4-partition-mac/
https://www.paragon-software.com/us/home/extfs-mac/#
https://apple.stackexchange.com/questions/210198/mount-ext4-on-el-capitan

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

alpew, 2021-01-04:
Thanks @starbasessd 
I've attached the following files:

Following 2 files before the Network Link switch was set to 'Off'

- working-boot.log (Configuring Wired Network set to 'dhcp')
- working-dmesg.log
[working-dmesg.log](https://github.com/ccrisan/motioneyeos/files/5766085/working-dmesg.log)
[working-boot.log](https://github.com/ccrisan/motioneyeos/files/5766086/working-boot.log)

The following 2 files are the logs immediately after the issue has occurred (reboot after the Network Link switch set to OFF) and can be confirmed by boot.log having the Configuring Wired Network set to 'No Link'

- boot.log
- dmesg.log
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5766081/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5766082/boot.log)

PsyEng, 2021-01-04:
@alpew No worry, it was just a hint for you :) 

One major difference between working/-dmesg.log are those lines:

working-dmesg.log
`[    1.294330]  mmcblk0: p1 p2`

`[    2.622262] hid-generic 0003:04D9:A088.0003: input,hidraw2: USB HID v1.11 Mouse [HID 04d9:a088] on usb-0000:01:00.0-1.4/input1`
`[   19.193638] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)`

`[   22.197199] random: smbpasswd: uninitialized urandom read (12 bytes read)`

`[   22.808155] i2c /dev entries driver`

dmesg.log
`[    1.291941]  mmcblk0: p1 p2 p3`

`[    2.622355] hid-generic 0003:04D9:A088.0003: input,hidraw2: USB HID v1.11 Mouse [HID 04d9:a088] on usb-0000:01:00.0-1.4/input1`
`[    2.655341] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)`
`[    3.039313] i2c /dev entries driver`

But this looks fine, because on first boot, it will create a new partition (p3), which takes a bit longer, then on other boots.

@starbasessd Do you now, where @alpew can change the option "Network Link Watch" at the filesystem?

If so, it would be good to see, what happens, if the setting is "restored".

starbasessd, 2021-01-04:
Not off the top of my head, @PsyEng, but give me a bit to check. Not sure
if it is a motionEye python script or motion feature...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PsyEng, 2021-01-04:
@starbasessd Thank you

starbasessd, 2021-01-04:
@PsyEng I found the python script that watches the network, but haven't had time to wade through it. Give it a go, and see if you see anything...
[watchctl.zip](https://github.com/ccrisan/motioneyeos/files/5766950/watchctl.zip)

PsyEng, 2021-01-05:
@starbasessd Nope, there's nothing...
But I meant, where he can change the settings back, because he doesn't have access to the webui. "os.conf", "prefs.json", or whatever.

starbasessd, 2021-01-08:
Sorry it took me a while to get back to this Real Life(tm) got me...
The file in question is /data/etc/watch.conf
The lines are "LINK_WATCH="true""
and "IP_WATCH-"true""
Those are the default 'normal' settings.
You should be able to ssh in and change them with nano.

teertsg, 2021-01-12:
I am experiencing exactly the same issue, I was using my Raspberry Pi 4 with a wired headless setup.

What I have found is that I am able to flash the image to a new SD card, add a new camera, make a few changes to the config, then when it reboots it is not available, I have moved it into another room with a screen and I have found that it says Configuring Wired Network: No Link.

I have tested this using the current version and dev version and am experiencing the same issue, I have re-flashed and got the same issue on many retries, nothing has changed to any of my original setup, its plugged into the same switch and I am using my DHCP server to give out a static IP address, etc. This config has been the same for the last two years or so.

The only way I am now able to run Motioneyeos is by setting it up in wireless mode, which I wanted to avoid as I want all my camera traffic segmented on a single network switch.

I hope you can get this resolved as this is now an issue for me too.

Thanks for your help!

starbasessd, 2021-01-12:
What kind of changes (other than adding/editing camera?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

teertsg, 2021-01-12:
In order to give you all the steps I have completed I will document all my steps below with a new SD card and a MacBook Pro with Big Sur 11.1:

1. Format SD Card using SD Card Formatter
2. Flash SD Card using Etcher with the image: motioneyeos-raspberrypi4-20200606.img
3. Plug SD Card into RPi 4 and switch it on.
4. Open Chrome and browse to http://192.168.1.102
5. Click message You have not configured any camera yet. Click here to add one... to add a new camera with following settings:
Camera Type: Network Camera
URL: rtsp://192.168.1.107/h264Preview_01_main
Username: xxxx
Password: xxxx
Camera: RTSP/TCP Camera
6. Change Timezone to Europe/London
7. Click Apply
8. Reboot System
9. Change Camera name to 'Front Camera'
10. Change Resolution to 800x600
11. Click Apply
12. Turn Off 'Text Overlay'
13. Change Movie Passthrough to 'On'
14. Change NTP Server to pool.ntp.org
15. Click apply
16. Reboot
17. Turn Off FTP Server
18. Turn Off Samba Server
19. Reboot
20. Increase Streaming Framerate to 15
21. Turn Off Network Link Watch
22. Reboot
Error Configuring Wired network: no link

So whilst documenting all the changes I have made for you, it looks like turning off Network Link Watch causes it to fail

In order to confirm I completed the following steps:

1. Format SD Card using SD Card Formatter
2. Flash SD Card using Etcher with the image: motioneyeos-raspberrypi4-20200606.img
3. Plug SD Card into RPi 4 and switch it on.
4. Open Chrome and browse to http://192.168.1.102
5. Click message You have not configured any camera yet. Click here to add one... to add a new camera with following settings:
Camera Type: Network Camera
URL: rtsp://192.168.1.107/h264Preview_01_main
Username: xxxx
Password: xxxx
Camera: RTSP/TCP Camera
6. Change Timezone to Europe/London
7. Turn Off Network Link Watch
8. Click Apply
9. Reboot
Error Configuring Wired network: no link

So it looks like I have identified the issue, in order to resolve it I copied my wpa_supplicant.conf to the root of the SD card

my wpa_supplicant.conf file is below:

country=US
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="SKY1088B"
    psk="REMOVED"
}

I restarted the Pi, connected to the WiFi address of 192.168.1.47 turned off WiFi, Turned On Network Link Watch and this resolves my issue.

I know I resolved the issue myself, but I wouldn't have if I hadn't documented all the steps in testing for you.

Is this expected behaviour?

Sorry for the long one...

starbasessd, 2021-01-12:
!st, was that you in the 3rd row of my last class on "how to give all the
info needed for issues"? Wow, thanks! (Seriously!)
2nd, should it work that way? No.
Why were you turning off the Network Watch setting? (Use Case?)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

teertsg, 2021-01-12:
I know what it's like when trying to troubleshoot issues, you then have all the info to replicate the issue or find I was doing something silly. :)

in answer to your question, No real specific reason, I was turning off anything which I felt I didn't need to reduce CPU usage to improve camera & motion performance as I have experienced some delays when using several cameras together and I have just purchased some new higher quality cameras.

starbasessd, 2021-01-12:
Network Watch serves several purposes ( and is a simple script)
Turning off all the other stuff makes sense, mostly, especially if you are
comfortable using scp to transfer pics and movies, or only wanting to use
the web browser to view and download.
I will leave this here and tag @ccrisan to see if it can / should be
changed slightly to take it into account (if Network Watch is off, then
don't force reboot if it doesn't see the Network Link up...)

One more datum, if you can / would, please: If Network Watch is off, does
the Ethernet still pull the DHCP issued Static IP address?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

teertsg, 2021-01-12:
Thank you for this info. 

I don’t think I can answer the final question because as soon as I turn off Network Link Watch and reboot I am unable to connect via Ethernet, my only option then is to use WiFi which then uses DHCP, but I have not configured a static address for the WiFi MAC. 

Let me know if you want me to conduct a test with anything specific, happy to help. 

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-12:
If you could, enable the wifi, connect the ethernet, and boot with Network
Watch turned off.
Then run
ifconfig
or
ip a
and check the Ethernet IP Address.
Does it show anything?
I would myself, but it's going to be this weekend before I can free one of
my Pi4s to test.

On Tue, Jan 12, 2021 at 11:04 AM teertsg <notifications@github.com> wrote:

> Thank you for this info.
>
> I don’t think I can answer the final question because as soon as I turn
> off Network Link Watch and reboot I am unable to connect via Ethernet, my
> only option then is to use [...]

alpew, 2021-01-13:
@teertsg

Hi. So it's not just me which is good to hear. 

Yes that is exactly my findings as well. Setting Network Link Watch to Off when MEOS is on a RPI4, will prevent the Pi from rebooting successfully getting a DHCP address from a network server and instead returns 'Configuring Wired Network: No Link' 

@starbasessd I have Wifi set to Off and the Pi picks up a wired DHCP address when Network Link Watch is left at the default 'on' position. 

So when Network Link Watch is turned off and the Pi reboots, The Pi will allocate itself an IPv6 Link local address (FE80) (ifconfig - see below) but no IPV4 just the 127.0.0.1 loopback address. 

[image]

starbasessd, 2021-01-13:
OK, built a Pi3B+, motioneyeos dev20201026
Test 1a Wired Ethernet only, no wpa_supplicant.conf file:
Boot normal, gets static IP address from DHCP, as expected.
Test 1b Wired Ethernet only, no wpa_supplicant.conf file, Expert Settings,
Network Link Watch OFF:
Boot  normal, gets static IP address from DHCP, as expected.
"Borrowed" grandsons Pi4B-4GB, EEPROM dated 3 Sep 2020 (latest critical)
Installed dev20201026 for RPi4
Test 2a Wired Ethernet only, no wpa_supplicant.conf file:
Boot normal, gets static IP address from DHCP, as expected.
Test 2b Wired Ethernet only, no wpa_supplicant.conf file, Expert Settings,
Network Link Watch OFF:
Configuring wired Network: No Link
No IP address listed in boot screen
ip a (console command line) reports <BROADCAST,MULTICAST,UP,LOWER_UP>
ifconfig eth0 reports flags=4163<UP,BROADCAST,RUNNING,MULTICAST>
Confirmed same behavior/result with Pi4 & 20200606

@ccrisan
Something is handling the turning off of Network Link Watch differently
from the Pi3B+ and Pi4. Instead of turning off watching the link and
resetting when necessary, it seems to turn off actively using it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Dozitreallymattr, 2021-01-13:
Same/similar problem on RP 3 B+

Trouble shooting included finding the "no link" message in the init.d file and changing the error messages from "no link" to "no link 1" and "no link 2" respectively.  Found that "no link 2" was where the system errored out.

I resolved this by modifying the S40network file init.d  I increased the sleep time for the net test see line below,  sleep was 1 increased it to 10.  DHCP's correctly each time.

while [[ "$(cat /sys/class/net/${OS_ETH}/carrier 2>&1)" != "1" ]]; do
        sleep 10

PsyEng, 2021-01-13:
>Something is handling the turning off of Network Link Watch differently
>from the Pi3B+ and Pi4. Instead of turning off watching the link and
>resetting when necessary, it seems to turn off actively using it.

Yes, the Pi4 has it´s own network chip, which starts late in the boot process, Pi3 and before using network over USB2.0, which is activated directly after booting.

> I resolved this by modifying the S40network file init.d I increased the sleep time for the net test see line below, sleep was 1 increased it to 10. DHCP's correctly each time.
> 
> while [[ "$(cat /sys/class/net/${OS_ETH}/carrier 2>&1)" != "1" ]]; do
> sleep 10
> 
> hope this helps

This solution should do it.
I remember the same issue in june last year, but thought it´s a feature, because there´s an option, which defines a panic-reboot, if the network is down for 10 or 20 seconds.

PsyEng, 2021-01-13:
https://github.com/ccrisan/motioneyeos/issues/2553#issuecomment-717399762

The solution was postet 1 day after the last dev-release...

alpew, 2021-02-09:
Solution posted by @PsyEng

closed 2021-02-09 by alpew