motioneye-project/motioneyeos #2613: Motion Eye Instal Debian Install

STUKguy, 2020-12-24:
Hi,

So i am trying to install Motion Eye on Debian 10 using this guide https://github.com/ccrisan/motioneye/wiki/Install-On-Debian but when i get to step 4 and run "pip install motioneye" i get this
[Motion.txt](https://github.com/ccrisan/motioneyeos/files/5740272/Motion.txt)

not sure what to do?

starbasessd, 2020-12-24:
What happens if you run the install instructions for Ubuntu 20.04 and Newer?
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-12-25:
@starbasessd please wait till Sunday before replying i dont want you to take time away from your family anf friends for this. Merry Christmas, hope your haing a joyfull one.
Yeah just tried that one and i still get the same error
[Motion2.txt](https://github.com/ccrisan/motioneyeos/files/5741976/Motion2.txt)

starbasessd, 2020-12-25:
Are you running all the commands as admin (either sudo ...  or  sudo su,
then the commands?)?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-12-25:
@starbasessd but sure why but i can only log in as "root" basically i have abandoned using the PI 4 like i was before a built a whole new system using a Dell Optiplex 970 with the Openmedia Vault OS installed on it which is based on Debian 10 i was hoping to install Motion Eye side-by-side with this, but failing that i can install it on portainer/docker and passtrough the camera's but im wondering if what you said previoudly will cause problems as both camera appear to be the same make/model

starbasessd, 2020-12-25:
I missed the part where you went with all new hardware and OS, sorry.
Still need to use the setup instructions for Ubuntu 20.04 and newer, even
on Debian. The upstream repos are getting really messed up with everyone
shutting down python2.7.
If you go the docker method, there is a pretty good docker image setup
instructions in the motionEye wiki. Just need to pass through the
/dev/videox devices when starting the container. Should be able to see them
as full OSs see the cameras better.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-12-25:
@starbasessd sorry should have said at the beging, but i had Open Media Vault OS running on a PI 4 which in the last year i had to rebuild 5 times and i just got fed up of it so im going back to hardware i am use to thats why i went to Dell Optiplex. I managed to install Motion Eye on docker 1 camera works fine but as soo as i plug in another one it does not come up and i get this
[image]

from what i can tell i added the correct cameras
[image]

Not sure why this other camera wont show

starbasessd, 2020-12-25:
On the host OS, run
lsusb
and
lsusb -t
and
dmesg | grep video
and report back.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-12-25:
@starbasessd 
[image]

starbasessd, 2020-12-25:
Doesn't seem to be mounting the cameras to /dev/video[x]
Can you install something that would use the cameras? If you have Desktop,
Cheese works (usually).
May not be quite as responsive for a while, pseudo-grandkids are catching
their 2nd wind (or the sugar is hitting their systems <grin>).

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-12-25:
@starbasessd I don't have access to the desk top only command line, what would you recommend? I will do this tomorrow lack of sleep is catching up with me

starbasessd, 2020-12-25:
OK.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-25:
Install and run (when you get a chance...)

sudo apt-get install v4l-utils
v4l2-ctl --list-devices

(assuming the cameras are attached)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-25:
/dev/video0 and /dev/video2 should work.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-12-25:
@starbasessd so I changed docker to this
[image]

But still nothing, restarted the container but still the same
[image]
[image]

starbasessd, 2020-12-25:
Did you delete and re-add both as local V4L2 cameras?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-12-25:
@starbasessd yep

starbasessd, 2020-12-25:
Try:

v4l2-ctl -d /dev/video0 --list-formats

and

v4l2-ctl -d /dev/video2 --list-formats

and post, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-25:
Well, at least Camera 0 doesn't support video compression (mjpeg) so is
going to suck USB bandwidth.
I found this:
https://superuser.com/questions/431759/using-multiple-usb-webcams-in-linux
and unless both cameras support some sort of compression, it's not going to
work...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-12-25:
@starbasessd I will take a look now, I thought I would unplug the camera that is working remove, readd, reboot docker and the camera worked but as soon as I plugged the other camera in removed, readd and rebooted only 1 camera workes

STUKguy, 2020-12-25:
@starbasessd read through that post and there was a section do with resources which got me thinking what if it was an issue with where the USBs were plugged in so powered down the pc plugged 1 camera in the front and 1 camera in the back went into docker removed, readd Ed and rebooted and both cameras worked. Going to leave it for a few days just to make sure it sticks

starbasessd, 2020-12-25:
Cool.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

STUKguy, 2020-12-31:
@starbasessd Thank you for your help again. im closing this issue everything has been working fine no issue what so ever. not sure why they both wouldnt work from the rear usb ports?/??

But thank you and have a great new year

closed 2020-12-31 by STUKguy