motioneye-project/motioneyeos #2614: how: zero as fast camera and NUC as motion/recording?

BETLOG, 2020-12-25:
Question:
Can I use multiple PiZeroW in fast network camera mode, and do the motion-detect and recording on a more powerful machine?

I'm trying to maximise the framerate the zero produces while still being able to capture motion events.

Initially I assumed it would be intrinsically set up to do that, but apparently pointing my browser to the hub machine simply alters the settings on the zeros/cameras, and I see no way to make the hub handle the load intensive things.

Am I missing something?

starbasessd, 2020-12-25:
What OS are you using on the NUC for the base?
What instruction set are you using for the install of motion/motionEye?
How are you adding the cameras?(what type of camera?
I use an i5TFF Lenovo for my daily production machine.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-12-25:
When I connect a PiZero/CSI camera to my i5 TFF, I have the PiZero set to Fast Network Camera (set in the WebGUI of the PiZero), and on the hub I connect via Network Camera (simple Mjpeg) 
Works great.

BETLOG, 2020-12-26:
> When I connect a PiZero/CSI camera to my i5 TFF,

Thats essentially what I was attempting.

> What OS are you using on the NUC for the base?

Kubuntu 18.04

> What instruction set are you using for the install of motion/motionEye?

NUC:

hmm, i was going to say:

`sudo -H pip install -U motioneye
`

But history says i did this, so maybe pip failed. To be honest I dont recall...and did have issues with the install instructions, but I want to say it was the sudo -H pip that was the solve (because i posted in another thread saying it was).

```
 1324  sudo -H pip install -U motioneye
 1325  cat /usr/local/share/motioneye/extra/motioneye.conf.sample
 1327  sudo apt-get install motion
```

  In any case, motioneye does work on the Nuc as evidenced by being able to load it's interface, and manipulate the zero settings through it... i just cant do that when the zero is in streaming mode.

PIZERO

`sudo ./writeimage.sh -d /dev/sdf -i "./motioneyeos-raspberrypi-20190911.img" -n 'ssid:psk'`

Yes, I do have actual values in the wpa-supplicant file-proper. Access via SSH/html isnt the issue.

> How are you adding the cameras?(what type of camera?

on the pi0 - its the cam with thezero-specific MMAL ribbon built in, so i just plug it into the connector and tell pizeroeye1 to use an MMAL camera via it's web interface, and set it to a stream mode (ive tried all the various permutations of that too)

https://www.aliexpress.com/item/1005001593475326.html

on the Nuc: same as you described: tell the nuc's motioneye web interface to add a non-local camera. I tried all 3 (net/remote/simple) and all the various port/dir combinations. That's why I cant see why it fails. It should be trivial. I can view/use the pizero camera normally, and the nuc can manage its settings, i just cant set the zero to any network stream mode and do still be able to connect to it on the nuc.

starbasessd, 2020-12-26:
Use the instructions for Ubuntu 20.04 and Newer here:
https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
Ubuntu/Debian are killing (have killed) upstream most of the python2.7
functionality especially pip2.

Correct for the PiZero (MMAL). For the motionEye, it'll be a Network or
Simple Mjpeg, using the URL: http://[ip_address]:8081
You turn on Fast Network Camera on the PiZeroW using the PiZeroW WebGUI,
which disables most other on-board funtionality.
On the NUC, you connect to the Network or Simple Mjpeg as above, and
control all setting there.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BETLOG, 2020-12-26:
> For the motionEye, it'll be

Yeah, this is what I did. That's why I'm baffled that it didn't work, no matter what combination of 'this-really-should-work's I tried. But after the issues I had with post-2019 versions I'm not really surprised.

I may try (as you suggest) reinstalling motioneye at a later time to bring multiple camera feeds into a pretty interface, but my search for an answer to this problem also led me to [RPi Cam Control](https://elinux.org/RPi-Cam-Web-Interface). Because it can easily produce high motiondetected framerates and resolutions onboard even a pizero; the need to offload motiondetection is now moot, and it is in fact more bandwidth efficient NOT to offload that anyway, so I plan to use it on all my zerocams.

closed 2020-12-26 by BETLOG