motioneye-project/motioneyeos #2618: Incomplete video file

quinctilius, 2020-12-28:
I have a number of video recordings that seem incomplete (i.e. no thumbnail, unplayable by any player I've tried)
I think the issue is a network one (I am saving the files on a network share) - That's my problem

Is there any way to salvage the video files/data?

starbasessd, 2020-12-28:
Which verion of MotionEyeOS?
Pi or other SBC?
Ethernet or wifi?
Where are you storing them? FTP or SMB server?
What kind of network connection?
What kind of cameras?
If you download them, can you play them with VLC?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

quinctilius, 2020-12-29:
It's version 20200606
Raspberry pi 3b
Wifi
Smb
Usb camera
No, can't play on VLC

Normally encoded as mp4. Have tried changing file names to avi for vlc to
fix, but didn't work

starbasessd, 2020-12-29:
This doesn't help for older files, but try switching to mp4 without OMX (or
any other video format without OMX). or switch to dev20201026.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-04 by starbasessd