motioneye-project/motioneyeos #2619: HELP PORT FORWARDING

chiragvshroff, 2020-12-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI Zero W

### Network Connection

My motionEyeOS unit is connected to the network via: wifi

STATIC IP OF MY PI IS 192.168.31.5
PLEASE CHECK ATTACHED IMAGES
QUESTIONS
1.Should i select tcp,udp or tcp and udp?
2.what should i put in external port
3.what should i put in internal port?

[image]
[image]

starbasessd, 2020-12-29:
1)  TCP
2) Anything you want, except 80. (Suggested: anything over 1024. Most
common are 8000, 8080, 10000) Depends on if you have anything else being
port-forwarded. Be aware, you are opening a major security hole in your
firewall, too.
3) If you are using motionEyeOS, 80. If you are using motionEye: 8765. If
you only want the video from a single camera:8081

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

chiragvshroff, 2020-12-29:
Thanks for the reply looks like my ISP has blocked port forwarding.

ivanjf1978, 2021-01-13:
I can not see motioneyeOS from outside my network internet.

I am using MotioeyeOS 20200606 in my Raspberry Pi 4B

I have tried everything and still not working. 

Please any help advice could be great. I know every router it is different but, someone could be say something useful 
for me for sure, I am not expert.

My internet provider says:

How to set up port forwarding

Open a new web browser window and delete the contents of the address bar.
Type 192.168.X.X into the address bar and press Enter. You'll see the Hub Menu.
Click the Advanced tab and enter the username and password.
 
Default username: XXXX
Default password: XXX or, your WIFI password.
If these have been changed, enter the current username and password.
 
Click LAN IP Setup then Add under Address reservation.
Choose the device the traffic should be forwarded to by clicking the appropriate radio button and take note of the IP it's assigned. This is a list of everything currently connected to the Hub. Switching off devices you don't want to set up port forwarding on will make the list smaller, helping you find devices by their Device Name.
Click Add. The router may request a restart. Click OK and allow the reset to complete before continuing.
Log back into the Hub if you need to. Go to Security, then click Services and Add Service.
 
If you have more than one port number / range (PS4 usually uses 4 numbers / ranges), repeat steps 8-11 for each making sure:

Each has a unique name (like PS4, PS4_1, PS4_2)
If you have a port number (like 443) you put this number in the Start AND finish boxes.
If you have a port range (like 443-445) you put the lower number in the start box and the higher number in the finish box.
 

Type a Name (something you'll recognise like PS4).
Choose the type or port (TCP / UDP or both).
Type the Start and Finish port.
Click Apply. If you see 'service name is already present in the existing firewall or custom services,' click OK.
Repeat from step 8 until you've added all port numbers/ranges, then continue.
Click the Firewall Rules.
Click Add under Inbound Services.
 

If you have a Q hub fill in the IPv4 settings.

Choose the name you set up from the Service dropdown.
Choose ALLOW always from the Action dropdown.
Next to Send to LAN Server type the IP address you noted earlier.
Click Apply to save your changes. If you added more than one port number/range go back to step 14 until you've added them all.
 
Why would I need to set up port forwarding?

The Hub (white) and Hub 2.0 (black) have a built-in firewall to prevent unauthorised external access to devices that are connected to the internet. To allow external access through your Hub (eg, online gaming or hosting of an FTP server), some software or applications may need to reconfigure the firewall. The Hub allows you to do this using a feature called Port Forwarding.
 
I’m having problems port forwarding – what should I do?

 First and foremost check:-

All of the ports you need are forwarded.
The right protocol is in use.
Any other active firewalls on your home PC (such as McAfee, Norton or Windows Firewall) may also need port forwarding set up.

My question is

I have did port forwarding in my router like mentioned before and it is not working

I have tried to do port forwarding on my PC with the windows 10 firewall in the part "Inbound roles"

Please, what it is the problem?

Any advice please?

Thanks

starbasessd, 2021-01-13:
For the Lan address: whatever IP address your Raspberry Pi uses
For the Lan Port: 80
For the outside port use: 8080
Use TCP or Both
Call it RPi
Follow the rest of the instructions.
Disconnect or turn off WiFi on your phone
Connect to the outside IP address (On a Windows or other computer, go to
website: http://whatismyip.com to get your outside IP address)
Should look like WWW.XXX.YYY.ZZZ
So you would go to http://WWW.XXX.YYY.ZZZ:8080
Good Luck.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2021-01-13:
Thank you very much @starbasessd 

Do I need to do port forwarding set up on Windows Firewall? I don't have screen on my Raspberry pi I am using my pc to watch my cameras on motionEyeOS 
>

starbasessd, 2021-01-13:
Not if you are trying to watch them from outside your home.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

aleraco, 2021-02-19:
you have to duckdns.org your IP. it has been my solution to get into motioneyeOS from outside my LAN.
due to the fact that every ISP changes your IP daily, night time for te most, there can't be any other solution to do that

starbasessd, 2021-02-19:
@aleraco Not true for ALL ISPs. I've had the same IP address for a long
time, since the last long term power outage in my area, which also took out
the Cable Internet Service.
I use a script on an internal Linux machine to check the Internet IP
address and email it every 2 hours, since before that they had been working
on upgrading my area's service to 200Mb/S and had changed a lot more often
than once a day.
I have the script in /boot/extip.sh:
-----
#!/bin/bash
myip=`wget -q -O - checkip.dyndns.org|sed -e 's/.*Current IP Address: //'
-e 's/<.*$//'`
sendemail -f [your_email_address] -o tls=yes -f [your_email_address] -t
[your_email_address] -s smtp.gmail.com:587 -xu [your_email_account] -xp
[your_email_password] -u "ExtIP" -m $myip
-----
It is run in a crontab (I use sudo crontab -e)
10 */2 * * *  /boot/extip.sh
This runs 10 minutes after the hour, every other even hour. If your ISP
changes more or less often, you can adjust the time appropriately.
The only 'extra' is sendemail, which is available in most, if not all
debian derived repositories (including RPi)  for a
sudo apt install sendemail
On my GMail, I filter by the subject line and double check if I seem to
have issues connecting.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-04:
If you are still having issues, please open a new issue.

closed 2021-03-04 by starbasessd