motioneye-project/motioneyeos #2621: USB Camera Setting/Viewing Settings discrepancies

vbbartlett, 2020-12-30:
### Preliminary Docs
I confirm that I have read the [CONTRIBUTING]
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

### Board Model
Raspberry Pi 4B 1,2,4 GB

### Camera
USB Cameras (Webcam/UVC)
https://www.amazon.com/gp/product/B08881Q6GZ/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B07Q2V27GK/ref=ppx_yo_dt_b_asin_title_o03_s01?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B088K566BK/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

### Network Connection
Wired
### Peripherals
None

### Log Files

When setting the video resolution and frame rate on these cameras, it TRIES but in the logs fails.  It then sets them to lower values (camera dependent) But the UI doesn't show this.  I set the camera to 1920x1080 and it would fail and set it to 640x360.  (see log extract below).  Could this be detected and somehow represented in the UI as a failure?  Showing RED?  Show both values.  What I found was that I could get the resolution that I requested AFTER adding "Extra Motion Options"  v4l1_palette 8 which sets it to mjpeg compression.   

So in the log below, it fails to set to 1920 and downgrades to 1280.  It also downgrades the fps from the requested 30 to 10.  Again in the UI I believe it worked but when I start viewing the videos or the images and they are not the quality requested, it is disconcerting.  Several of the cameras drop all the way to 640x360. Spend several hours trying to get it to work before realizing that Motion was downgrading it and there was nothing I could do about it... at least without specifying other options.  
(Side note: frustrating that these cameras work on window at 1080p/30fps but not on Pi4, not a MotiopEye failing)  

`[1:ml1:Second] [DBG] [VID] v4l2_device_capability: ------------------------
[1:ml1:Second] [DBG] [VID] v4l2_device_capability: cap.driver: "uvcvideo"
[1:ml1:Second] [DBG] [VID] v4l2_device_capability: cap.card: "UVC Camera: UVC Camera"
[1:ml1:Second] [DBG] [VID] v4l2_device_capability: cap.bus_info: "usb-0000:01:00.0-1.1"
[1:ml1:Second] [DBG] [VID] v4l2_device_capability: cap.capabilities=0x84A00001
[1:ml1:Second] [DBG] [VID] v4l2_device_capability: ------------------------
[1:ml1:Second] [NTC] [VID] v4l2_device_capability: - VIDEO_CAPTURE
[1:ml1:Second] [NTC] [VID] v4l2_device_capability: - STREAMING
[1:ml1:Second] [DBG] [VID] v4l2_input_select: Name = "Camera 1", type 0x00000002, status 00000000
[1:ml1:Second] [NTC] [VID] v4l2_input_select: Name = "Camera 1"- CAMERA
[1:ml1:Second] [NTC] [VID] v4l2_norm_select: Device does not support specifying PAL/NTSC norm
[1:ml1:Second] [NTC] [VID] v4l2_pixfmt_select: Configuration palette index 17 (YU12) for 1920x1080 doesn't work.
[1:ml1:Second] [NTC] [VID] v4l2_pixfmt_select: Supported palettes:
[1:ml1:Second] [NTC] [VID] v4l2_pixfmt_select: (0) MJPG (Motion-JPEG)
[1:ml1:Second] [DBG] [VID] v4l2_pixfmt_select: 0 - Motion-JPEG (compressed : 1) (0x47504a4d)
[1:ml1:Second] [NTC] [VID] v4l2_pixfmt_select: (1) YUYV (YUYV 4:2:2)
[1:ml1:Second] [DBG] [VID] v4l2_pixfmt_select: 1 - YUYV 4:2:2 (compressed : 0) (0x56595559)
[1:ml1:Second] [NTC] [VID] v4l2_pixfmt_select: (2) H264 (H.264)
[1:ml1:Second] [DBG] [VID] v4l2_pixfmt_select: 2 - H.264 (compressed : 1) (0x34363248)
[1:ml1:Second] [NTC] [VID] v4l2_pixfmt_set: Testing palette YUYV (1920x1080)
[1:ml1:Second] [WRN] [VID] v4l2_pixfmt_set: Adjusting resolution from 1920x1080 to 1280x720.
[1:ml1:Second] [NTC] [VID] v4l2_pixfmt_set: Using palette YUYV (1280x720)
[1:ml1:Second] [DBG] [VID] v4l2_pixfmt_set: Bytesperlines 2560 sizeimage 1843200 colorspace 00000000
[1:ml1:Second] [NTC] [VID] v4l2_pixfmt_select: Selected palette YUYV
[1:ml1:Second] [INF] [VID] v4l2_fps_set: Trying to set fps to 30
[1:ml1:Second] [INF] [VID] v4l2_fps_set: Device set fps to 10
[1:ml1:Second] [INF] [VID] v4l2_ctrls_list: ---------Controls---------
[1:ml1:Second] [INF] [VID] v4l2_ctrls_list:   V4L2 ID   Name and Range
[1:ml1:Second] [INF] [VID] v4l2_ctrls_list: ID09963776 Brightness, 0 to 100
[1:ml1:Second] [INF] [VID] v4l2_ctrls_list: ID09963777 Contrast, 0 to 100
[1:ml1:Second] [INF] [VID] v4l2_ctrls_list: ID09963778 Saturation, 0 to 100
[1:ml1:Second] [INF] [VID] v4l2_ctrls_list: ID09963779 Hue, 0 to 100
[1:ml1:Second] [INF] [VID] v4l2_ctrls_list: ID09963788 White Balance Temperature, Auto, 0 to 1
[1:ml1:Second] [INF] [VID] v4l2_ctrls_list: --------------------------
`

sebasptsch, 2021-02-28:
I'm having this issue as well. I know that my usb webcam supports 1080*1920 recording although every time I set it, it reverts back to 860*420.

starbasessd, 2021-02-28:
More than one USB camera connected at the same time?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vbbartlett, 2021-02-28:
@starbasessd yes I have each of my Rasp Pi 4s has two usb cameras attached and streaming.

starbasessd, 2021-02-28:
On my Pi4-4GB and Pi4-8GB, I find the dev20201026 works much better all
around.
You can do a backup prior to either an in-place upgrade (enable
pre-releases in Expert Settings, then check for update)
or install a clean image and do the restore of settings.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sebasptsch, 2021-02-28:
No, this happens when only a single usb camera is attached and the v4l2
palette in the extra args does fix it in the end.