motioneye-project/motioneyeos #2622: How send picture and video to telegram when trigger motion

inkdude, 2020-12-31:
I want to send picture (screenshot of video) when triggering motion

I use section
 * Run an end command

 * `curl -s -X POST "https://api.telegram.org/botXXX/sendPhoto" -F chat_id="XXX" -F photo=%f`
But it's doesn't work
can anybody help with this?
[image]

starbasessd, 2021-01-07:
You may want to put your command in a bash script and see if it works there. If it does, then call the command here. Also, you may need to put a delay of 1-2 seconds [sleep 2] befor the command to give motionEye a chance to create the file.

inkdude, 2021-01-09:
I use this and it's works perfectly

https://github.com/DaniW42/motioneye-telegram

closed 2021-01-09 by inkdude

skizzo, 2022-02-22:
Anybody's got an idea how to transmit the name of a created movie file?