motioneye-project/motioneyeos #2624: I can not add Outdoor camera Onvif in motioneyeOS

ivanjf1978, 2021-01-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version:

motionEyeOS 20200606

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

Raspberry PI 4B 

### Camera

I am using the following type of camera: Network Camera

My camera model is: Yoosee 1080P Full HD & 355°Pan/tilt

This is the camera

https://www.amazon.co.uk/gp/product/B08HGNQP9V/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
Information of the camera Wifi

Specification:
Software： YOOSEE interesting
System structure： embedded Linux system, the structure of ARM chips
Chip：AK3918E (nk)
Pixel：200 w
The sensor resolution：2.0 M "GS2265 1920 x1080 by one-third
Lens：4 mm
Viewing Angle： 80 °
PTZ yuntai adjust Angle： horizontal, vertical direction of 355 ° ： 90 °
Can preset a： six
Video compression standard： h. 265/20 FPS
Video formats: PAL
Minimum illumination: 0.01 Lux @ (F2.0, VGC ON), O.L uxwith IR
Electronic shutter： automatically
Digital dynamic： wide support
Noise reduction：2 d and 3 d
Infrared lamp： four white six infrared lattice lamp Network connection：wi-fi wireless transmission (support IEEE802.11 b/g/n wireless protocol)
Night vision：Double filter automatic switching, 5-10 meters (owing to the different environment)
White light：APP end control：(1) open (2) shut down (3) automatically
(to be automatic mode, the IR - open automatically after the CUT CUT night vision infrared lamp, intelligent detection of the human body, intelligent opening/closing incandescent lamp)
Audio：Built-in microphone and speakers, omnidirectional noise reduction support two-way real-time audio transmission,
ADPCM audio compression standard, the adaptive streaming
Call the police: (1) motion detecting, push support images
(2), artificial intelligence researchers intrusion detection (optional)
ONVIF： Support ONVIF
Storage： TF card (maximum 128 g)： Cloud storage/cloud disk (optional)
The power supply. 12 v / 2 a
Working temperature：- 10 ℃ ~ + 50 ℃ work humidity: 95% RH or less)

The camera asked to create password to connect any device with the camera, see picture

I can see the streaming video on VLC using the password that I mentioned before

rtsp://admin:password@192.168.0.21:554/onvif1

Please, why I can not add this camera to my motioeyeOS?

thanks in advance 

Regards

Ivan

starbasessd, 2021-01-02:
did you try rtsp://[ip_address/onvif and then put user and password in the
appropriate boxes?
how about rtsp://[ipaddress]/11 ?
These were the only two for Yoosee cameras at
https://www.ispyconnect.com/man.aspx?n=Yoosee

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2021-01-02:
Thank you for your email

I tried all this:
rtsp://admin:password@192.168.0.21:554/onvif1 (I can see the streaming
video on VLC with this format. This format it is not suitable on
motioneyeos)

rtsp://192.168.0.21:554

rtsp://192.168.0.21:554/onvif1

And not working

I will try like you say

Thank you very much

Regards

ivanjf1978, 2021-01-03:
Hi @starbasessd 

I have tried these all and not working:

http://admin:password@192.168.0.21/image.jpg

rtsp://admin:password@192.168.0.21/onvif1

rtsp://admin:password@192.168.0.21/11

http://admin:password@192.168.0.21/cgi-bin/snapshot.cgi?chn=554&u=admin&p=password

http://admin:password@192.168.0.21/snap.jpg?JpegCam=554

starbasessd, 2021-01-03:
DO NOT use the admin:password in the rtst URL. PERIOS

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2021-01-03:
I have tried as well, but not working

http://192.168.0.21/image.jpg <http://admin:password@192.168.0.21/image.jpg>
>
> rtsp://192.168.0.21/onvif1 <http://admin:password@192.168.0.21/onvif1>
>
> rtsp://192.168.0.21/11 <http://admin:password@192.168.0.21/11>
>
>
> http://192.168.0.21/cgi-bin/snapshot.cgi?chn=554&u=admin&p=password
<http://admin:password@192.168.0.21/cgi-bin/snapshot.cgi?chn=554&u=admin&p=password>
>
> http://192.168.0.21/snap.jpg?JpegCam=554
<http://admin:password@192.168.0.21/snap.jpg?JpegCam=554>

ivanjf1978, 2021-01-04:
have tried as well, but not working

http://192.168.0.21/image.jpg 

 rtsp://192.168.0.21/onvif1 

 rtsp://192.168.0.21/11

 http://192.168.0.21/cgi-bin/snapshot.cgi?chn=554&u=admin&p=password

 http://192.168.0.21/snap.jpg?JpegCam=554

Please, what can I do? Whats wrong?

Thanks

Ivan

Joanvoo, 2021-01-05:
maybe 
rtsp://192.168.0.21:5000/onvif1        (udp)
rtsp://192.168.0.21:5000/onvif2        (udp)
rtsp://192.168.0.21:554/onvif2          (tcp)

ivanjf1978, 2021-01-05:
Thank you very much

Let me try

Regards

Ivan

ivanjf1978, 2021-01-05:
What about the password to access to the camera? The camera can not allow
leave it without password and with the password I can access to the camera
throw any device

Joanvoo, 2021-01-05:
put user and password only in the
appropriate boxes from motioneyeOS
[image]

starbasessd, 2021-01-07:
Also please try to upgrade to dev20201026.

ivanjf1978, 2021-01-09:
Hi, I have tried all the ways that I can. and I can see from camera 3 it is all in grey. The cameras 1 and 2 I keep it swift off thats way are in grey as well like the camera that I trying to add.

[image]

ivanjf1978, 2021-01-09:
I tried to use "ONVIF device manager" I not streaming live video coming from this camera. And I can see the URL address, but not working on MotionEyeS

[image]

starbasessd, 2021-01-09:
Out of curiosity, what resolution do you have the camera itself set to? My
ReoLink I can force a resolution (I use 1280x720, but if I force it higher
or to native, it fails.)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2021-01-09:
In the camera It has 3 options; HD, SD and LD. I put SD

[image]

ivanjf1978, 2021-01-09:
I can see streaming video on VLC using this URL

rtsp://admin:pass@192.168.0.21:554/onvif1

[image]

ivanjf1978, 2021-01-09:
From the camera's APP I press enable to connect and password otherwise the camera can not work without password 

[image]

ivanjf1978, 2021-01-09:
On MotionEyeOS all the options are in grey colour and when I I set up video 320x240 I cane see

[image]

ivanjf1978, 2021-01-09:
> rtsp://192.168.0.21:5000/onvif2

[image]

[image]

ivanjf1978, 2021-01-09:
[image]

[image]

ivanjf1978, 2021-01-09:
ALLLLEEELLLLLUYYYYYYAAAAAAAAAAAAAAAAAAAAAA

Thank you very much @Joanvoo 

[image]

I didn't notice to change in the bottom part on MotioEyeOS

RTSP/TCP or RTSP/UDP

I was using TCP when I changed for UDP and......wahalla

Working

Thank you very much all guys that you help me. I was thinking to throw it way my new camera or my Raspberry Pi hahahahha

starbasessd, 2021-01-09:
That camera is apparently not compatible with the onvif or rtsp
implementations used by motionEye or motionEyeOS through motion & ffmpeg..
Nothing suggested by myself or other users has helped you.
Many cameras are like that. Many cameras use a proprietary or obscure
version of those protocols that VLC and some browsers can work around, but
motionEye and motionEyeOS use motion & ffmpeg for stream manipulation, and
either of those may not support your camera.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2021-01-09:
It working now @starbasessd 

I didn't notice to change in the bottom part on MotioEyeOS

RTSP/TCP or RTSP/UDP

I was using TCP when I changed for UDP and......wahalla

ivanjf1978, 2021-01-09:
Thank you very much @starbasessd for your help and support ;-)

closed 2021-01-09 by starbasessd

djeev1, 2022-08-05:
Hi  All

After trying a lot, thanks to all the shares in these posts, I've managed to get my Yoosee HD MINI WIFI Camera to work with MotionEye. It is important to change to RTSP/UDP at the last selection choice when creating the camera. THanks also to @ivanjf1978 for sharing a screen capture.