motioneye-project/motioneyeos #2625: Upredictible motion detection behaviour

RaD0, 2021-01-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

motionEye Version 0.42.1
Motion Version 4.2.2+gitUNKNOWN
OS Version motionEyeOS 20200606

### Board Model

Raspberry Pi 4 with 4GB Ram and 64GB SD card

### Camera

I am using the following type of camera: 

1. IP Camera TP-Link Tapo C200 (https://www.tp-link.com/in/home-networking/cloud-camera/tapo-c200)
2. IP Camera CP Plus (https://www.cpplusworld.com/cp-unc-ta21pl3)

### Network Connection

One camera with WiFI and another PoE

### Issue

Both the cameras are connected to a Raspberry Pi 4 over WiFi and Ethernet. The issue I'm facing is with motion detection. It does not seem to behave in an expected similar way. Sometimes small motion is detected and sometimes a very obvious motion (like someone moving) is not detected. Sometimes a motion gets detected at one instance, but when done at a later point, it doesn't get detected. For example: It gets detected when I walk by the camera. But no detection happens when I walk by again an hour later. The issue I'm facing is hard to explain since it doesn't behave in a predictable manner. I turned off the auto settings, so that the settings remain static all the time. The following screen shot is the motion detection settings. Please suggest how to tune my settings: 

TP-Link WiFi Camera
[image]

CP Plus PoE Camera
[image]

starbasessd, 2021-01-07:
Pllease upgrade to dev20201026 and see if that improves your issue.

RaD0, 2021-01-08:
I'll try it out. Thanks!

closed 2021-03-04 by starbasessd