motioneye-project/motioneyeos #2626: No live camera image after update

pkwentus, 2021-01-04:
Hello-

When I log into Motioneye the live camera feed does not show up- I have the spinning wheel  for each camera with icons above to review photos etc.   Is this a setting issue?

Raspberry Pi 3B+
motionEye Version | 0.39.3
-- | --
Motion Version | 4.1.1
OS Version | motionEyeOS 20190119

Cameras are wired network type using RTSP to provide images. 
Camera Name: IPCAM
Serial Number: VVVIPCNTD161201Z-18870e1vIf4gd0C
Hardware Version: 8700-sc2135
Software Ver: V2.3.4.2103-S50-NTD-B20170508B

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-01-04:
Please update to something that isn't 2 years old, then report.
Please attach the messages and motioneye logs.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pkwentus, 2021-01-04:
I'm using MotioneyeOS and I've always updated via the user interface.   Are you suggesting re-imaging the SD card from a fresh build?   You are correct- my implementation is a few years old... 

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5766197/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5766198/motion.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5766199/messages.log)

starbasessd, 2021-01-04:
Absolutely. Your image is probably too old to upgrade via the user
interface (too many changes).
You're overheating or under voltage (Power Supply too small)
Jan  4 12:14:54 meye-3d56eb58 user.notice throttlewatch: currently:
arm-frequency-capped
Jan  4 12:15:24 meye-3d56eb58 user.notice throttlewatch: currently:
arm-frequency-capped
Jan  4 12:15:54 meye-3d56eb58 user.notice throttlewatch: currently:
arm-frequency-capped
Jan  4 12:16:24 meye-3d56eb58 user.notice throttlewatch: currently:
arm-frequency-capped
Jan  4 12:16:54 meye-3d56eb58 user.notice throttlewatch: currently:
arm-frequency-capped
Newer images (20200203 or newer can help diagnose it)
I recommend 20200606 or dev20201026 found here:
Releases · ccrisan/motioneyeos
<https://github.com/ccrisan/motioneyeos/releases>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pkwentus, 2021-01-04:
Well first- thank you.  

Secondly  I did set this up a few years back and "just used it".  Motioneye performed well for me and this became somewhat of an appliance.   Obviously it is quite easy to re-image a SD card and apply a backup file.  Perhaps this should be a good "best practice" every few years ;)  

As to my heat issue- the Pi WAS in a case and the log files didn't lie.  I appreciate you pointing this out and I'll keep an eye on it going forward.

Cheers!

closed 2021-03-04 by starbasessd