motioneye-project/motioneyeos #2627: motion eye os to play an mp3/ wav file

Hazzoid, 2021-01-05:
Hi 

I don't have any coding knowledge but I am currently setting up some raspberry pi's as cctv camera's and I am wanting the one at the door to play a small mp3/ wav file when it detects motion and i was wondering how I would go about this.

cheers 
Hazzoid

starbasessd, 2021-01-05:
It would have to be anything but a PiZero (No audio output)
You could use aplay triggered by the Run A Command function in Motion
Notification
I have not tried it personally, but it is intriguing... I may post
something later if I can find some time to test...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-05:
What hardware are you using? Pi2, 3, 3b+, 4?
Which version of motionEyeOS are you using?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-05:
I built a Pi3B+ with dev20201026 on a 4GB SDCard (no intention of recording
to it)
I copied a wav file (doorbell.wav) to /boot, along with my
wpa_supplicant.conf file and ssh.txt file before 1st boot
I inserted and booted to it
I then set up 3 network IP cameras.
In Motion Detection, set the cameras the same as my production Camera Hub
for detection
in Motion Notification, I enabled Run A Command and entered:
aplay -D hw:0,0 /boot/doorbell.wav
and saved
(BTW, it's wav because it appears there are no mp3 codecs available by
default in motionEyeOS)
Now, when it detects motion, it plays the sound through the headphone jack
(there isn't any HDMI attached)
At this time I don't have an extra monitor with sound to test with.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Hazzoid, 2021-01-06:
I am currently using a pi 3

starbasessd, 2021-01-06:
Same image.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-04 by starbasessd