motioneye-project/motioneyeos #2629: Txt extension

Duffy1963, 2021-01-07:
I can't get my wpa supplicant file right. I have not been able to remove the .txt extension and unable to see it. The file appears with the .conf extension but the pi just keeps looking for a wired connection. I have tried the last few versions of the OS and all were the same. I am disabled and accessing the wired connection is very problematic. Anyone who can help will be my hero. I have been using a raspberry 3zero w (I verified the w) and man do I  wish we had these little wonders back when I was in the military. ### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-01-07:
Make sure you are NOT using Notepad in Windows to create/edit the file.
Use Notepad++ or other Linux-friendly text editor.(also helps with CR/LF vs
CR only line editing)
Use Command Prompt to rename the file from wpa_supplicant.conf.txt to
wpa_supplicant.conf
In Windows File Explorer, click on view at the top, then put a check mark
in "File Name Extensions" toward the right.
Let me know

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Duffy1963, 2021-01-07:
Thanks, you were a big help. I am now up and running. Sent from my Galaxy

starbasessd, 2021-01-07:
Per requestor, issue resolved

closed 2021-01-07 by starbasessd