motioneye-project/motioneyeos #2630: Won't install on a brand new Raspberry Pi 4 2G

123g-man, 2021-01-07:
I've been using Motioneyeos on a raspberry pi 3b without issues. I have just bought a new Pi4 2G board with a view to upgrading. Re-flashed the SD card with the Pi4 20200606 version, plugged in and powered up. Install not working. Have attached a LCD screen - display says;

recover4.elf not found
start4.elf is not compatible
This board requires newer software
Get the latest software from https://raspberrypi.org/downloads/
ERROR 00000044

it keeps going round this loop. Ive re-flashed the board with the latest Raspberry Pi OS via Raspberry Pi Imager, ive re-flashed the sd card with Raspberry Pi Desktop and it runs ok, so the board and sd card are both ok. Downloaded motioneyeos again in case of bad download and re-flashed the sd card again - same result. 

Any ideas?

starbasessd, 2021-01-07:
With the Raspberry Pi OS installed, run the commands

sudo apt update
sudo apt full-upgrade
sudo reboot

Then run

sudo rpi-eeprom-update -a
sudo reboot

 and then run
vcgencmd bootloader_version

It should say
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000

and Rpi4 (my 4GB & 8GB both) will boot with 20200606 and dev20201026.
I recommend the latter, there are 'quirks' with 20200606.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

123g-man, 2021-01-07:
Hi,

Thank you for the quick reply. I checked and the Pi did have the latest firmware Sep  3 2020 13:11:43. I tried the dev20201026 and this works without problems! Great - problem solved!! Must be some bug in 20200606 .

Many thanks for your help

G

starbasessd, 2021-01-07:
One of many. Thanks for the update. Please close if resolved.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

123g-man, 2021-01-07:
Hi , all was going well but my Pi has just died. Hopefully an unrelated issue!

Regards,  G

closed 2021-01-07 by 123g-man