motioneye-project/motioneyeos #2631: Remote login trouble

Duffy1963, 2021-01-07:
I have successfully made the wpa_supplicant file work and the username "admin " got the zero up and running. I am new to remote logins and can't seem to make it work. I have tried from an android phone and also a Windows 10 laptop. I guess I am missing some key point. Help. ## Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-01-07:
What kind of remote login?
What kind of error are you getting?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Duffy1963, 2021-01-07:
I got the pi zero running and since it appears to have no gui I can fire up, I am trying to connect from another computer to configure motioneyeos.  I hope I explained that well,  I  am  a little out of touch with this  technology. Thanks for responding.Sent from my Galaxy

starbasessd, 2021-01-07:
You open a web browser (preferably not Chrome or Edge) and go  to the IP
address: http://[ip_address]
It will prompt for username, which is admin, and password leave blank.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Duffy1963, 2021-01-07:
Sorry I forgot to mention that all I get is "this site can't be reached 127.0.0.1 refused connection"Sent from my Galaxy

starbasessd, 2021-01-07:
What is the IP of the PiZero?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Duffy1963, 2021-01-07:
When I type hostname -i it tells me 127.0.0.1 although I am dealing with the default font on a seven inch monitor so maybe the 0's are 8's.  Nope they are not 8's.Sent from my Galaxy

starbasessd, 2021-01-07:
Try:
ip a [enter]
or
ifconfig \enter\
If a PiZeroW look for wlan0

Duffy1963, 2021-01-07:
You are a steely eyed rocket man. Thank you for all your help and I look forward to being able to help others down the road.Colin Winterburn CD1 Sent from my Galaxy

starbasessd, 2021-01-07:
So that got you in?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Duffy1963, 2021-01-07:
Sure did. Thanks again.Sent from my Galaxy

starbasessd, 2021-01-07:
Per user, corrected and working.

closed 2021-01-07 by starbasessd