motioneye-project/motioneyeos #2632: Raspberry Pi Zero W won't recognise wireless setup.

davehiggins21, 2021-01-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi Zero W
### Camera

I am using the following type of camera: PI NOIR with integrated IR lisghts

My camera model is: Pi NOIR

### Network Connection

My motionEyeOS unit is connected to the network via: WiFI

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 The PI starts to boot but then displays 'confuring wired network and displays Panic - rebooting in 60 seconds caused by network

The supplicant.conf file is formatted as below 
country=GB
update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={scan_ssid=1
ssid="VM6931"
psk="rnqyjgs9hs"}

The file is in the root of the SD card which I used the recommended tools to format and create. I've tried to work out where I'm going wrong but and tried changing the file multiple times but it just won't boot. Any help would be very welcome.

Thanks Dave

starbasessd, 2021-01-08:
Example wpa_supplicant.conf file contents and format (with your info):
-----
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=GB

network={
    ssid=" VM6931"
    psk=" rnqyjgs9hs"
    key_mgmt=WPA-PSK
}
-----
Do not use notepad to edit, use a linux friendly editor like Notepad++
Make sure, on the back of the Pi under the holes for the GPIO it says
Raspberry Pi Zer0 W V1.1
If it says Raspberry Pi Zero V1.3 it does NOT have wifi built in.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

davehiggins21, 2021-01-09:
Many thanks for that. I was using the instructions in the WIKI which are  as below. Your format worked first time so thanks for that.

Dave

Create a new file in the ~30MB part of the disk that you can still access on Windows called supplicant.conf. This file tells the Pi how your network is set up and will read as follows:

country=US

update_config=1

ctrl_interface=/var/run/wpa_supplicant

network={

            scan_ssid=1

            ssid="your network's SSID"

            psk="your network's password"

}

 

To be clear, you will need to put quotation marks around "SSID" and "password," as shown in the text above. Also, US indicates that your device is in the United States. This and other country codes are based around the ISO 3166-1 alpha-2 standard, which may be relevant if your Pi will reside elsewhere.

closed 2021-01-09 by davehiggins21

starbasessd, 2021-01-09:
And the filename must be

wpa_supplicant.conf

not just supplicant.conf

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

davehiggins21, 2021-01-09:
Yes I'd spotted that from your email. Can the wiki be corrected because there may be others finding this problem and giving up.

Dave

starbasessd, 2021-01-09:
I have requested the changes. I have access to motionEye and motionEyeOS
documentation, but that pieve is in thingOS, and I don't (currently) have
access there.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?