motioneye-project/motioneyeos #2633: Nothing from /boot/ is loaded --> /boot/config.txt of wpa_supplicant.conf

ovz93br43v7, 2021-01-09:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi-20200606.img

### Board Model

I am using the following board/model: Raspberry Pi Zero W v1.1

### Camera

I am using the following type of camera: Raspberry Pi NoIR Camera V2

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: 5 inch LCD (800 x 480)

I have problems to get anything loaded from the folder /boot/.
As described here [https://github.com/ccrisan/motioneyeos/issues/2578](https://github.com/ccrisan/motioneyeos/issues/2578) already found a solution for the not loading of /boot/wpa_supplicant.conf

But I would like to use my HDMI 5 inch 800 x 480 display.
This is my config.txt file which I successfully tested with RaspberryPiOS lite (HDMI working, LED off):
```
dtparam=act_led_trigger=none
dtparam=act_led_activelow=on
overscan_right=48
hdmi_group=2
hdmi_mode=14
```
source HDMI options: https://github.com/ccrisan/motioneyeos/issues/485
source LED options: https://github.com/ccrisan/motioneyeos/wiki/Turn-Off-LEDs

I already tried the following things:

1. to format and re-flash the motioneyeos image several times with three different sd cards without success. (I use elemtaryOS [ubuntu] and formater and image burner form Appcenter)
2. I tried the newest pre-release image of motioneyeos
3. I tested my config.txt with RaspberryPiOs lite --> worked immediately
4. I tried to edit the file with gedit and with nano (terminal)

I am absolutely at a loss and any ideas are welcome. :-(

starbasessd, 2021-01-09:
Since you are using a Linux, are you putting the files / editing the files
in the Linux /boot or the SDCard /boot?
If I run
df -h
my motionEyeOS boot folder shows as /dev/mmcblk0p1 in the first column, and
where it's mounted in the last column
for example  /media/my_user/FA36-CDBA
so I would edit
sudo nano /media/my_user/FA36-CDBA/config.txt
and then eject the disk and insert into the PI and boot.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-09:
Forgive my stupidity, I always tried to put this file to /media/username/uuid-of-sd-card/boot and not to the small boot partion...

closed 2021-01-09 by ovz93br43v7

ovz93br43v7, 2021-01-09:
Everything is working for now! Thank you a lot!
During start up I see the following message on my tiny LCD:
```
Starting streameye: /data/etc/streameye.conf: line  9: unexpected EOF while looking for matching `
/data/etc/streameye.conf: line 10: syntax error: unexpected end of file
/data/etc/streameye.conf: line 9:  unexpected end of file while looking for matching ``'
/data/etc/streameye.conf: line 10: syntax error: unexpected end of file
```
Should I do anything to prevent this error?

starbasessd, 2021-01-09:
where did that file come from? on my clean motionEyeOS, that file doesn't
exist.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-09:
I have checked my sd card, content of this file is:
```
PROTO="mjpeg"
PORT="8081"
RTSP_PORT="554"
MJPEG_WIDTH="640"
MJPEG_HEIGHT="480"
MJPEG_FRAMERATE="5"
MJPEG_BITRATE="800000"
AUTH="disabled"
CREDENTIALS="user:long-password-with-special-chars:motionEyeOS"
```

starbasessd, 2021-01-09:
Does your last line use the character `  or ' in the password?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-10:
You are right that was the reason.