motioneye-project/motioneyeos #2634: RPi 4 2GB - No video feed from USB camera and external WiFi dongle does not work

pratapj81, 2021-01-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

Yes, I confirm

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

Yes, I confirm

### motionEyeOS Version: 

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi4 Model B 2GB RAM

### Camera

I am using the following type of camera: V4L2 

My camera model is: [RunCam USB camera](https://www.banggood.in/RunCam-WebCam-1-or-2_9-inch-COMS-1920+1080-1080P-FOV105-Full-HD-USB-FPV-Camera-p-1712493.html) 

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * Ubiquiti Wifistation USB Adapter 802.11g/n up to 1000mW
 * Edimax EW-7811Un 

### Log Files

I consider the following log files relevant to this issue:

[boot.log](https://github.com/ccrisan/motioneyeos/files/5792743/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5792744/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5792745/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5792746/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5792747/motioneye.log)

Hello,

I am seeking help for the following two issues: 

1. RunCam USB camera does not work with my RPi 4. I only see a 'no camera' icon in the MotioneyeOS GUI irrespective of whether I use the built in WiFi or the Ubiquiti Wifistation USB Adapter (which doesn't work either. See next issue). 
2. An external dongle, such as the Ubiquiti Wifistation USB Adapter or the Edimax EW-7811Un  does not work even after setting the `OS_WLAN` to `wlan1` in the /data/etc/os.conf file. 

Both these peripherals worked fine with ver 20200606 of MotioneyeOS on a Raspberry Pi B+

Things I have tried to fix these issues:

1. Used a step-down buck and powered the RPi 4 directly from the GPIO pins. This did not help. 
2. Blacklisted brcmfmac in the modprobe.conf file by following the instructions from [here](https://kerberosio.zendesk.com/hc/en-us/articles/115001013585-Use-WIFI-dongle-on-the-Raspberry-Pi-3). This did not help either. 
3. Installed MotionEye on Raspbian by following the instructions [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian). The USB camera worked only after I ran `sudo apt install fswebcam`.  Out of the box, Raspbian sees both WiFi cards and shows me wlan0 and wlan1 in the menubar. But my first preference is to run MotionEyeOS on the Pi. 

The location where I intend to setup the MotionEye camera is very far from the my router. So, it is important that I use an external WiFI adapter such as the Ubiquiti Wifistation USB adapter. I also intend to use the CSI camera (NoIR) and the USB webcam, connected to the same board. This is why I upgraded from the Raspberry Pi B+ to the Raspberry Pi 4B 2GB. In my initial tests with the Raspberry Pi B+, where I setup and ran MotionEyeOS for a few days, the Ubiquiti Wifistation, RunCam USB camera and Edimax EW-7811Un all worked fine. However, the WebUI was sluggish, WiFi connectivity poor, and overall experience very disappointing. I attributed this to the Raspberry Pi B+ and bought the Raspberry Pi 4B 2GB but am disappointed that both the USB camera and the external cards don't work. I am yet to receive the Arducam NoIR CSI camera from the vendor. 

I have attached the log files for the USB camera issue here. Since I don't have connectivity after setting OS_WLAN` to `wlan1` in the /data/etc/os.conf file, I am attaching screenshots from my monitor: 

[image]
[image]

If you need any more information which will help fix this issue, please let me know. Thanks in advance for your help!

starbasessd, 2021-01-10:
Please try dev20201026 on the Pi4B-2GB, as there are some unresolved quirks
with 20200606 on the Pi4s.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pratapj81, 2021-01-10:
Tried dev20201026 but facing the same issues. Attached are the logs. 

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5792895/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5792896/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5792897/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5792898/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5792899/motion.log)

pratapj81, 2021-01-11:
I'd like to report that the RunCam USB camera works if I unplug it from one USB port and plug it into another, after MotionEye starts. No luck with the external WiFi adapter yet.

pratapj81, 2021-01-13:
Documenting the process I followed to get my RPi4 and MotioneyeOS up and running: 

1. Insterted the Pi4 microSD card to a SD card reader and installed Raspbian Lite OS from the Raspberry Pi [website](https://www.raspberrypi.org/software/operating-systems/). I used a MacOS system running MacOS 10.14.6. 
2. Unplugged the microSD card from the SD card read, and plugged it back in to my Mac. 
3. Created a blank **SSH** file with no extension and copied to 'boot' partition using the Mac Finder. 
4. Copied the **wpa_supplicant.conf** file to the boot partition using the Mac Finder. 
3. Connected the Pi4 to a monitor via a microHDMI cable. 
4. Put the microSD card back into the Pi4 and booted it. 
4. Logged in to the Pi4 via SSH and Terminal on the Mac by typing `ssh pi@raspberrypi.local` and `raspberry` as the password. 
5. Ran the following commands to upgrade the packages: 
`sudo apt update -y`
sudo apt upgrade -y
`sudo reboot`
6. Changed the timezone by following the steps listed [here](https://peppe8o.com/change-timezone-in-raspberry-pi-os-lite-from-terminal/). 
5. Installed Motioneye by following the steps given [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian). Used `sudo` before **every** command. 
6. Logged in to the Motioneye webUI by typing `http://[your_ip]:8765/` in the browser. 
7. Connected the external dongle and camera to the USB ports. Saw a low voltage warning, in spite of using a 3A power supply for the Pi. So, connected the dongle and camera via a powered USB Hub, which was in turn connected to the Pi4. 
8. Unplugged and plugged USB camera back again and got the feed. 
8. Disabled on-board WiFi the Modprobe Blacklist method listed [here](https://pimylifeup.com/raspberry-pi-disable-wifi/). 
9. Gave my Pi4 a static IP by following the instructions listed [here](https://thepihut.com/blogs/raspberry-pi-tutorials/how-to-give-your-raspberry-pi-a-static-ip-address-update). 

Currently, testing this setup for reliability.

starbasessd, 2021-01-13:
As to motionEyeOS:

    Used a step-down buck and powered the RPi 4 directly from the GPIO pins. This did not help.
Not a good idea, as it bypasses all on board and internal voltage and current protections in the PI. 

    Blacklisted brcmfmac in the modprobe.conf file by following the instructions from here. This did not help either.
When you plug in the Ubiquity wifi dongle and boot motioneye, what does the command
ifconfig [enter]
show? You can access command line via console or ssh.

    Installed MotionEye on Raspbian by following the instructions here. The USB camera worked only after I ran sudo apt install fswebcam. 
If Raspbian doesn't support the camera 'out of the box', then the odds are highly unlikely that motionEyeOS will, either. Since there are no processes to use apt, or apt-get, or dpkg within motionEyeOS, it is somewhat difficult to install packages. Firmware isn't mych easier, but usually only involves a folder or two that can be copied into place. (ZD1211 wifi dongles for example)
Out of the box, Raspbian sees both WiFi cards and shows me wlan0 and wlan1 in the menubar. But my first preference is to run MotionEyeOS on the Pi.
Which 'both'? Onboard and Ubiquity, onboard and EDI, or Ubiquity and EDI?

closed 2021-03-04 by starbasessd