motioneye-project/motioneyeos #2635: No still images captured after update

anarch9791, 2021-01-11:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS ver: 0.42.1

### Board Model

I am using the following board/model: Raspberry PI 3B+.

### Camera

I am using the following type of camera: MMAL, 
My motionEyeOS unit is connected to the network via: WiFi, 

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
After updating rasbian my pi cam stopped capturing still images. I can see live feed but when trying to capture still it is not recording it. I'm using standard build with no attachments or modifications. Is there any command or config file I would have to change to reboot still images capture? Again camera is working as I can see live feed but unable to capture still image. I get the blanc screeen with unable to open video device,

starbasessd, 2021-01-11:
Disk Free?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-11:
MotionEye or motionEyeOS?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

anarch9791, 2021-01-11:
20GB of free space and it is motionEYE installed on raspbian

starbasessd, 2021-01-13:
Please post in the motionEye github issues:
https://github.com/ccrisan/motioneye/issues

closed 2021-01-13 by starbasessd