motioneye-project/motioneyeos #2636: Unable to apply 'Remote' motionEye camera settings.

jamesbrindle, 2021-01-11:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: **dev20201026**

### Board Model

I am using the following board/model: **Raspberry PI 4**

### Camera

I am using the following type of camera: **MMAL**

My camera model is: **Pi Cam Module v2.1**

### Network Connection

My motionEyeOS unit is connected to the network via: **Ethernet**

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Message

Hi, I apologise if this is duplicate. I've got a good look, but can't find anything related to my query...

I'm using your build: dev20201026 rather than the previous 20200606 stable version as the 20200606 version seems not to be unable to pick up the Pi camera module v2.1? (no option of MMAL and V4L2 just shows a grey picture) However the dev20201026 does so I'm using that (only works will MMAL though).

I am attempting to use my **Pi 4** as the 'hub' and I have another camera on a Raspberry 3B+ using the stable 20200606...

My issue is that although I can add the 'Remote motionEye' camera to the 'hub' and view it, I seem to be unable to change any settings of the camera. I just get this message below when I try to 'Apply':

[image]
&nbsp;

I'm just wondering if this is a known issue or is there something wrong with my setup?

FYI, amazing work BTW!!

Kind regards,

J

starbasessd, 2021-01-11:
All kinds of weird little quirks for 20200606. Try upgrading the Pi3 to
either dev20201026, or downgrading back to 20190911.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jamesbrindle, 2021-01-11:
OK sure, will give that a try... I don't suppose there's a way for force the Pi 20200606 to update to the dev20201026 release is there, or do I need to re-flash the SC card?

Thanks for responding.

starbasessd, 2021-01-11:
In Expert Settings, Turn on Dev Releases.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jamesbrindle, 2021-01-11:
Ha, of course there is... Sorry

jamesbrindle, 2021-01-11:
That's completed worked! Thanks for your help!

closed 2021-01-11 by jamesbrindle