motioneye-project/motioneyeos #2637: RasPi4 Boot loop

pineapplee85, 2021-01-11:
I am running motionEyeOS version: motioneyeos-raspberrypi4-20200606

### Board Model

I am using the following board/model: Raspberry Pi 4 2GB

### Camera

I am using the following type of camera: MMAL

My camera model is: Cheap Night vision fisheye from amazon (Previously worked fine on rpi 2b with motioneyeos.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi, ethernet for testing if need be
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

### Log Files

I consider the following log files relevant to this issue:

Hi there,
When I try to boot up the Pi it gets stuck in a boot loop with 4 flashes of the activity light (4 slow-ish then 4 fast)

Any advice would be helpful. I can't hook up a display as I don't have the right adapter sorry about that.

Thanks,
Ethan!

starbasessd, 2021-01-11:
First, install RaspberryPiOS on the SDCard.
Run
sudo apt update && sudo full upgrade -y [enter]
reboot
sudo rpi-eeprom-update -a
reboot
vcgencmd bootloader_version [enter]
If it reports anything older than Sep 3 2020 then run through it again.
Once that is updated, please re-install motionEyeOS from:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz
and let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pineapplee85, 2021-01-11:
I get 

sudo: full: command not found

starbasessd, 2021-01-11:
Sorry, typo:
sudo full-upgrade -y

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pineapplee85, 2021-01-11:
sudo: full-upgrade: command not found

haha my pi is not behaving today

starbasessd, 2021-01-11:
Not you. Interrupted by grandson..
sudo apt full-upgrade -y

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pineapplee85, 2021-01-11:
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTLOADER: up-to-date
CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

starbasessd, 2021-01-11:
Should now be able to get anything after 20200203 to boot, and dev20201026
is the best choice.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pineapplee85, 2021-01-12:
Thanks,
Will try tomorrow and update soon

pineapplee85, 2021-01-12:
That did the trick! Thank you very much 👍

pineapplee85, 2021-01-12:
Hi,
Worked fine for a little bit (an hour or so) but then suddenly the pi went off the network. Now it is just stuck in the boot loop again. Not sure what to do now :/

starbasessd, 2021-01-12:
What other settings changes did you make in the process of setting it up
for motionEyeOS in motionEyeOS?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pineapplee85, 2021-01-12:
I installed motioneyeos, set up the wifi connection, added my camera (fisheye night vis- set it to just under 1080p res @30 fps), smb storage, cont recording with file lengths of 1800 seconds, no motion detect

starbasessd, 2021-01-12:
Which version of motionEyeOS and how connected? Did you change version as
suggested, and are you wifi'd or ethernet or both?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pineapplee85, 2021-01-12:
Yes, used the motioneyeos-raspberrypi4-dev20201026 and was using wifi when the  boot loop issue began. (internal adapter)

starbasessd, 2021-01-12:
I'm not going to be able to reproduce until this weekend.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

123g-man, 2021-01-12:
I had exactly the same problem with a brand new Pi 4 2G.  I was using 20200606 and it kept booting. Changed to dev20201026 and it ran OK -  for about an hour, then the Pi completely died! Sent back to the PiHut for replacement. New board runs OK with dev20201026 and has been running without issue for several hours. If your Pi 4 is new then perhaps there is a bad batch?

pineapplee85, 2021-01-12:
> I'm not going to be able to reproduce until this weekend.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

No problem, anything I can do to help in the meantime?

> I had exactly the same problem with a brand new Pi 4 2G. I was using 20200606 and it kept booting. Changed to dev20201026 and it ran OK - for about an hour, then the Pi completely died! Sent back to [...]

This pi is not new, I was using it for a satellite receiving station which it ran fine. That's interesting that it died through, what kind of cooling solution did you have?

starbasessd, 2021-01-12:
@123-gman 's comment is interesting.

To test, I'd run Raspberry PiOS on it for a while, and see if you get
similar results...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

123g-man, 2021-01-13:
Hi

> This pi is not new, I was using it for a satellite receiving station which it ran fine. That's interesting that it died through, what kind of cooling solution did you have?

I didn't have any heat sinks fitted, just the board but in open air - not in  a case. The new Pi 4 continues to work without issue.

G

pineapplee85, 2021-01-15:
Hi again,
Slight little update: I seem to have resolved the boot loop issue. I upgraded the power supply and cable and there doesn't seem to be an issue (so far)
I do however, get a jumpy video. I'm running at 1280x1024 @ 30 fps which is high, but thought the system would have been able to handle that?
Its always jumpy when streaming live, but also comes out jumpy when looking back at the recordings on the smb server.

Thanks!

starbasessd, 2021-01-15:
Depending on what SBC you are using (and even within a family like the Pis,
they are not powerful enough to try to do 30fps at those resolutions. The
software does the best it can, within the limitations of the hardware, and
what is requested.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pineapplee85, 2021-01-19:
Hi, I have tried it on this resolution and it seems to be working okay, however suddenly it will hang on an image and stop recording. Might try lower the fps and see what happens from there.

starbasessd, 2021-03-04:
More than 30 days no input

closed 2021-03-04 by starbasessd