motioneye-project/motioneyeos #2639: Login password problem

jraff9, 2021-01-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

jraff9, 2021-01-13:
Yesterday I down loaded the "motioneyeos-raspberrypi4-20200606.img.xz" file and flashed it to a USB card. Installed it in my Raspberry Pi 4 and attempted to boot it. I was presented with all sorts of configuration text followed by a "Login".
Ok where is the doc on the login? and I assume there is a password also, where? HELP!  What file do I need to read?

closed 2021-01-13 by jraff9

starbasessd, 2021-01-13:
Hopefully you menat microSDCard, not USB
Username is "admin" (no quotes, all lower case)
password is just [enter] (no password
There are instructions on that screen, it tells you how to get to the management screen, as there is very little you can do on the console.

jraff9, 2021-01-13:
Thanks, it worked. Now I have to get a camera attached. Working on that.
Where are the setup instructions?

starbasessd, 2021-01-13:
https://github.com/ccrisan/motioneyeos/wiki

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jraff9, 2021-01-13:
The WEB interface is working. Now I have to figure out what camera it senses and how to configure it.

starbasessd, 2021-01-13:
https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios

Most, if not all options have a little button to the right, that if you
hoverr your mouse over it, will explain what the option is looking for, or
what it needs, or what it will do.
If you want specific answers, post a new issue, anser the questions in the
post, and we'll try to help. "How do you set it up" is not a specific
question, as there is so much that can be done.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jraff9, 2021-01-13:
Thanks for the help. Play time starts now! Bye!