motioneye-project/motioneyeos #2642: Shutdown via Terminal, SSH

Klausoleum, 2021-01-14:
How can I shut down the system via SSH?

MotionEyeOS does not know about shutdown
with poweroff nothing belongs

starbasessd, 2021-01-14:
halt
BTW, no need to use "sudo", you are already root.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-04:
No update for > 30 days

closed 2021-03-04 by starbasessd