motioneye-project/motioneyeos #2645: Mounting filesystems: mount: mounting /dev/mmcblk0p3 on /data failed: Input/output error

jorijnsmit, 2021-01-17:
- motionEyeOS Version: `20200606`
- Board Model: Raspberry Pi Zero W
- Camera: MMAL
- Network Connection: pre-configured Wi-Fi

Installed onto 16GB card using `sudo ./writeimage.sh -d /dev/disk2 -i motioneyeos-raspberrypi-20200606.img -n 'ssid:pass'`. On first boot takes some time formatting disk, all successful. Can connect to Wi-Fi and get access to the panel from my laptop through port 80. Camera works 'out-of-the-box'. Awesome!

However, when rebooting, the following error pops up:
```
---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: mount: mounting /dev/mmcblk0p3 on /data failed: Input/output error
failed
 * Mounting overlay filesystems: mkdir: can't create directory '/data/usr': Read-only file system
```
It goes on with errors like this all based on the fact that `/data` does not exist. The boot procedure does not finish properly, I cannot ping it on its IP, no SSH, no port 80.

What is this input/output error?

### Log Files

```
/Volumes/NO NAME $ ls -al
total 26236
drwxrwxrwx@ 1 jorijnsmit  staff    16384 Jan  1  1980 .
drwxr-xr-x  5 root        wheel      160 Jan 17 13:41 ..
drwxrwxrwx  1 jorijnsmit  staff     2048 Jan 17 13:42 .fseventsd
-rwxrwxrwx  1 jorijnsmit  staff    24201 Jun  6  2020 bcm2708-rpi-b-plus.dtb
-rwxrwxrwx  1 jorijnsmit  staff    23938 Jun  6  2020 bcm2708-rpi-b.dtb
-rwxrwxrwx  1 jorijnsmit  staff    23719 Jun  6  2020 bcm2708-rpi-cm.dtb
-rwxrwxrwx  1 jorijnsmit  staff    24379 Jun  6  2020 bcm2708-rpi-zero-w.dtb
-rwxrwxrwx  1 jorijnsmit  staff    52304 Jun  6  2020 bootcode.bin
-rwxrwxrwx  1 jorijnsmit  staff      129 Jun  6  2020 cmdline.txt
-rwxrwxrwx  1 jorijnsmit  staff      112 Jun  6  2020 config.txt
-rwxrwxrwx  1 jorijnsmit  staff     9816 Jun  6  2020 fixup.dat
-rwxrwxrwx  1 jorijnsmit  staff  2134635 Jun  6  2020 fwupdater.gz
-rwxrwxrwx  1 jorijnsmit  staff  2142357 Jun  6  2020 initrd.gz
-rwxrwxrwx  1 jorijnsmit  staff  5155912 Jun  6  2020 kernel.img
drwxrwxrwx  1 jorijnsmit  staff    16384 Jun  6  2020 overlays
-rwxrwxrwx  1 jorijnsmit  staff  3794600 Jun  6  2020 start.elf
```

starbasessd, 2021-01-17:
Sounds like a glitch in writing the image. Please try re-imaging, or using the dev20201026 or 20190911 images.

jorijnsmit, 2021-01-17:
Tried re-imaging multiple times but always the same problem occurred.

Switching to `dev20201026` did fix the problem for me!

closed 2021-01-17 by jorijnsmit