motioneye-project/motioneyeos #2646: [Question] Why is my camera picture blue when I use Zoom X/Y?

ovz93br43v7, 2021-01-17:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: Fast Network Camera
My camera model is: MakerHawk Raspberry Pi Camera IR Fisheye Wide-angle 150-160 Degree 5MP OV5647 Webcam Automatically Switching between Day-Vision and Night-Vision Shooing Mode for Raspberry Pi

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: none

Normally the picture with IR cameras are with a light red tone, like this when I don't use the Zoom X/Y function of motioneyeos:
[image]

When I use the Zoom X/Y function to make the picture better centred, I get a strong blue colour in the picture:
[image]

I already tried several things to change in the settings but I can't find anything to prevent the blue colour. When I disable the Zoom X/Y then the colour switched back to the light red tone.

Is there any setting which could help me to remove the strong blue tone?

starbasessd, 2021-01-17:
Please try dev20201026 image

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-18:
Ok, is it possible to install this dev image with the internal updater?

starbasessd, 2021-01-18:
In Expert Settings, Enable Dev Images, then check for update.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ovz93br43v7, 2021-01-27:
I updated successfully to the latest dev version, but didn't changed anything --> picture is still blue.