motioneye-project/motioneyeos #2647: Sudden Wifi Issues

sectshun8, 2021-01-19:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

Have tried various versions...

### Board Model

Raspberry PI 3B+

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi, though Wired for initial setup

So I've been using motionEyeOS since late 2018 and never had any issues until now.  I work offshore and this allows me a nice way to keep tabs on my home when I'm away.  About halfway through last month, I noticed my cams had gone offline.  When I got home, I tried connecting locally via the wifi and I could not connect to my devices, even though my router showed they were indeed connected to the network.  Could not ping them either.

I have two cameras, both on Pi 3B+ models, and running the newest update, 20200606.  I reflashed both, hooked up via a wired connection for initial setup, no problem.  Entered in my usual wifi details, no issues. As soon as I removed the wired connection, and move the device to it's end location and power backup... nothing.  Again, my router can see its connected to my wifi network, but I can't actually connect to it via the web interface.

I've tried reverting back to a version from 2019, no luck.  

Oddly enough, if I leave both wifi and wired connections on DHCP, they each of course get their own individual IPs... but I can't access the web interface on either.  As soon as I disabled the wifi connection, I can then access the interface on the wired.  But then RPi goes into a panic reboot initiated by the wifi going down.  Which then puts me back to square one.

I've reflashed and only used a Wifi connection, never a wired and again, the wireless is recognized by the router, the RPi gets a DHCP address, and then again, I cannot connect via the web interface.

In all of this, if I never use the wifi, the wired connection works flawlessly.  Really wierd whatever is going on... be strange if both RPi's hardware physically went bad at the same time.

Any ideas?

starbasessd, 2021-01-19:
Update to dev20201026 or downgrade to 20190911.
Does the Pi exhibit the same issues if running RaspberryPiOS?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-01-19:
I have tried both the new dev available and the September 2019 release... both exhibiting the same symptoms.

And I have a third RPi running RaspberryPiOS that is having no issues with the Wifi at all, running my Plex media server wonderfully.  Though it is a RPi 4, whereas the two having issues are 3B+'s.

starbasessd, 2021-01-19:
> I have tried both the new dev available and that 2019 release... both

Running the RaspberryPiOS on the 2 Pis showing symptoms would find out if
the PIs are failing, or it's a mEOS issue.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-01-19:
Can confirm with RaspberryPiOS on the two devices in question, using the same wpa_supplicant.conf file on boot works exactly as it should, no issues.  Devices were given the same DHCP IP's as they were assigned using mEOS.  Connects to the internet, but oddly... I can ping my other RPi devices, but not my Windows machines.

starbasessd, 2021-01-19:
By name or ip address or both?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-01-19:
Both... I generally always ping by IP, but yeah, hostname doesn't work either.  But like I said, with all three RPi's on the network, including my Synology Disk Station... I can ping between of them by IP just fine.  And come to find out, my one Windows laptop upstairs can ping the RPi's as well... just the not the one I'm currently working on.  Were you thinking a DNS issue?

starbasessd, 2021-01-19:
try
tracert [ipaddress]
from the windows boxes to the Pis,
and
traceroute [ipaddress]
from the PIs to the windows boxes, and google .com

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-01-19:
Neither the Pi's or the Windows machine can trace to each other, both timeout, host unreacheable.  Both can route to google.com with no issue.

Both the 2 questionable Pi's, as well as the Windows machine can ping my 3rd Pi (media server) with no issue.  And the media server can ping the others as well.

starbasessd, 2021-01-19:
Check the settings in your DHCP server. Especially the Gateway.
Make sure the gateway is the same subnet, and the actual router. If you
have 2 (one for Wired and 1 for wifi, make sure they all point to the wired
router/modem/whatever.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-01-19:
All the network setups are the same, except for the actual IPs.  All are on the same subnet, same gateway, etc.  I had at first thought maybe because my laptop wasn't "Discoverable" on the network... but that was not the issue.  I can ping my both ways between my media server Pi and my Windows laptop no problem.

starbasessd, 2021-01-19:
If traceroute/tracert can't find the other end, there is no route.
Unless you set the addresses to static, and mis-set the gateways, or the
subnets are set up incorrectly, they should be able to see each other.
for example:
One of my Pi test hubs is 192.168.200.5/24 (ip: 192.168.200.5. subnet
255.255.255.0. gateway/gw: 192.168.200.1. dns:192.168.200.1, 8.8.8.8)
My PC is at 192.168.200.250/24 (ip: 192.168.200.250. subnet 255.255.255.0.
gateway/gw: 192.168.200.1. dns:192.168.200.1, 8.8.8.8)
both can ping each other and the gateway and the google.com
My dhcp server always issues the same IP address and other info from a list
of known MAC addresses.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-19:
My route on the PIs:
Destination     Gateway         Genmask         Flags Metric Ref    Use
Iface
default         router2         0.0.0.0         UG    202    0        0 eth0
default         router2         0.0.0.0         UG    303    0        0
wlan0
192.168.200.0     0.0.0.0         255.255.255.0   U     202    0        0
eth0
192.168.200.0     0.0.0.0         255.255.255.0   U     303    0        0
wlan0

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-01-19:
So my RPi's are:  192.168.0.XX and 192.168.0.XY
Windows PC: 192.168.0.XZ
Subnet: 255.255.255.0
Gateway: 192.168.0.1
DNS: 194.168.8.100, 8.8.8.8

I have a pretty simple setup, I've checked these over time and time again... and nothing in them has changed from when they were working, to now.  No router changes, no device changes, nothing.  And again, if I don't use the wpa_supplicant file, and just start straight away with a wired connection, BAM, no issues, straight in.  As soon as I turn the wifi on in the software with my credentials and reboot... can no longer access the interface wired or wireless.

starbasessd, 2021-01-19:
No clue, then. There is a setup issue.
When both Pis have both interfaces up, there should be 4 ip addresses
issued (one each to both ethernets, and one each to both wifis.

When you tried the RPiOS on the Pi3s, and both the wifi and ethernet are
up, you can ssh into both ip addresses of the same PI?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-01-19:
So from the RPi 4 running RaspberryPiOS, I can SSH into both the wired and wireless connections on the mEOS RPi.  Likewise, I can ping back and forth between them.  I can also ping the gateway, the DNS, and a few other devices on my network... just not the PC I use to view the cameras.

However, with both RPi 3B+'s with mEOS and RaspberryPiOS respectively, I cannot ping between the two on either wired or wireless IP, let alone SSH.  But I can ping the gateway and other devices except the two RPi units to each other and them both to my Windows PC.  There's deifnitely something specific to those two boxes that is causing this.

I had thought maybe my Powerline network was interfering, but have since ruled that out as well.  Still soem wierd stuff going on... surprised it's happening on freshly flashed units.

I'll get to the bottom of it eventually.

starbasessd, 2021-01-19:
Can you post your wpa_supplicant.conf with the ssid and pass phrase
redacted?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Sharpie, 2021-01-20:
I'm seeing a similar issue trying MotionEyeOS on a Raspberry Pi Zero W 1.1 for the first time. The device boots and connects to the network, but the connection is very flaky. If I'm lucky, I can get a SSH session after a half-dozen tries but the connection will destabilize and close whenever I try to do anything more complicated than `cd` or `ls`, such as `dmesg` or `less /var/log/messages`.

Re-flashing the SD card to use `2021-01-11-raspios-buster-armhf-lite.img` instead of `motioneyeos-raspberrypi-20200606.img` allows the hardware to maintain a stable network connection.

The `wpa_supplicant.conf` I added to `/boot` for MotionEyeOS is from [this template in the wiki](https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration):

```
country=US
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
  scan_ssid=1
  ssid="<redacted>"
  psk="<redacted>"
}
```

And the `wpa_supplicant.conf` added to `/boot` for Raspberry Pi OS is from [this template in the Pi docs](https://www.raspberrypi.org/documentation/configuration/wireless/headless.md):

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
 ssid="<redacted>"
 psk="<redacted>"
}
```

starbasessd, 2021-01-20:
Interesting.
Mine is:
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="ssid"
    psk="passphrase"
    key_mgmt=WPA-PSK
}
Which works in both mEOS and RPIOS across all devices ( even pre-wifi built
in Pis with added dongles )
I'll update the docs. Thanks for the pointer.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

rdukes6514, 2021-01-20:
> So my RPi's are: 192.168.0.XX and 192.168.0.XY
> Windows PC: 192.168.0.XZ
> Subnet: 255.255.255.0
> Gateway: 192.168.0.1
> DNS: 194.168.8.100, 8.8.8.8
>
> I have a pretty simple setup, I've checked these over [...]

The DNS you have listed here is the WAN side of your router, right?
And your using your router for your internal DNS/DHCP?
Make sure your router's DHCP setup is using itself as the primary DNS server...
Double check all  the numbers - I don't know how many times I've used an o instead of a 0!

rdukes6514, 2021-01-20:
Oh, and reboot everything...

sectshun8, 2021-01-20:
@starbasessd my wpa_supplicant file is exactly as yours, but with GB as my country instead of US (I've tried US as well with no luck).

As for using my router itself as it's own DNS, that's not an option I have.  I'm using what has come from my ISP in router mode.  I supposed I could use it in modem mode and use a separate router for the network... but really that's not going to happen. 

Everything has been restarted from the router to each of the client machines.  Let's not forget here, the RPI's are visible as connected to the router.  I can ping the router from each.  I can ping internet sites from each... it's just the talking back and forth to other clients on the network that's the issue.  And everything is setup as DHCP, so all auto assigned for each of the clients, no typos ;) (I have tried static as well but didn't help).

And it WAS all working, like literally last week still working.

starbasessd, 2021-01-21:
It's possible that your ISP thumb-fingered a setting. It happens. Have them check it out, describing the issues you're having. Have them check your DHCP service on the router, make sure it's set up correctly.

Sharpie, 2021-01-26:
I took another try at setting things up from scratch over the weekend and WiFi on the Pi Zero worked this time around. So, whatever was causing my initial troubles is not re-producing its self.

sectshun8, 2021-01-26:
> 
> 
> It's possible that your ISP thumb-fingered a setting. It happens. Have them check it out, describing the issues you're having. Have them check your DHCP service on the router, make sure it's set up correctly.

So my ISP did something that only effects the two RPi's on my network... but not the other RPi, or the 3 PC's, or the phones, tablets, tv's... just the two RPi's?

sectshun8, 2021-01-26:
So a new observation today.  Just trying to get back tko square one, a fresh mEOS install, no wpa_supplicant pre-loaded, running on a wired connection, DHCP.  No issues.  Can bring up web interface on my Windows PC, can ping back and forth, no issues.  Now as soon as I enable the Wifi with correct details and also DHCP and reboot... I can no longer communicate with the RPi via the Windows PC... not even on the wired IP.  So it's directly related to enabling/using the wifi on the RPi.  Subsequently disabling the Wifi does not restore the Wired connection and just makes the unit panic restart.

starbasessd, 2021-01-26:
2 defective Pis?
Using the wrong image on the wrong PI?
On the back of the PiZeros is a silkscreened line that says either
Raspberry Pi Zero V1.1 or something else.
On the front, under the GPIO pins, it will tell you any other specific
model fp Ip it is.
Which are they?
If you install RaspberryPiOS on them, preferably Lite, and only enable
wifi, are you able toreproduce the issue?
I have, in front of me now, a
PiZeroW (Wireless a/b/g/n wifi built in),
motioneyeos-raspberrypi-dev20201026.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz>
motioneyeos-raspberrypi-20200606.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi-20200606.img.xz>
Pi1 (no wifi built in, using a wifi dongle),
motioneyeos-raspberrypi-dev20201026.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz>
motioneyeos-raspberrypi-20200606.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi-20200606.img.xz>
Pi2 (also no built in wifi, using a dongle)
motioneyeos-raspberrypi2-dev20201026.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi2-dev20201026.img.xz>
motioneyeos-raspberrypi2-20200606.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi2-20200606.img.xz>
Pi3B (Wireless a/b/g/n, but not 5GHz ac wifi built in)
motioneyeos-raspberrypi3-dev20201026.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz>
motioneyeos-raspberrypi3-20200606.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi3-20200606.img.xz>
Pi3B+ (wifi up to ac wireless) motioneyeos-raspberrypi3-dev20201026.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz>
motioneyeos-raspberrypi3-20200606.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi3-20200606.img.xz>
Pi4B 4GB (with up to ac wireless) motioneyeos-raspberrypi4-dev20201026.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz>
motioneyeos-raspberrypi4-20200606.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi4-20200606.img.xz>
Pi4B 8GB (with up to ac wireless) motioneyeos-raspberrypi4-dev20201026.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz>
motioneyeos-raspberrypi4-20200606.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi4-20200606.img.xz>
Using Win32DiskImager and / or Raspberry Pi Imager on Windows 10, copying
the same wpa_supplicant.conf file to all images 30MB /boot partition.
I will report boot status of each shortly, (relatively) and none will be
with ethernet.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-01-26:
On another note of observation, on the RPi's in question where once I enable the wifi the connections no longer work... I read another thread in regard to dongles and just happened to have one.  I put it in the USB, configured ap1 with an IP and BAM... the wifi works over the USB dongle with no issues, even comes up on my Windows PC and Phone.

sectshun8, 2021-01-26:
And both of my RPI's are:
Raspberry Pi 3 Model B+

Near the GPIO are the letters JB.

And yes I'm using the correct release for my board.

starbasessd, 2021-01-26:
motioneyeos-raspberrypi3-dev20201026.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz>
or
motioneyeos-raspberrypi3-20200606.img.xz
<https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi3-20200606.img.xz>
???
Observationally you've discovered the software works for wifi, just not
with the onboard. If it works with mine, then I'd bet that it's an issue
with your PIs onboard wifi chips, maybe they overheated or otherwise failed.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-01-26:
I've tried both... as both of the ones listed should work for my board.

Question about the dev releases... how can you actually get into the settings on those in the web interface?

starbasessd, 2021-01-26:
Image 8GB microSDCard with Win32DiskImager, using dev20201026 and my
standard wpa_supplicant.conf file and empty ssh.txt file
boot log:
-----
Using username "admin".
Welcome to meye-f7e06d23!
[root@meye-f7e06d23 ~]# cat /var/log/boot.log
---- booting motionEyeOS dev20201026 ----
 * Detecting disk device: /dev/mmcblk0
 * Creating data partition: done
 * Formatting data partition: done
 * Mounting data partition: done
 * Copying data skeleton: done
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Setting empty root password: done
 * Copying version file: done
 * Setting UTC timezone: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: none
 * Starting wpa_supplicant: done (brcmfmac)
 * Configuring wireless network: dhcp
 * Setting current date using SNTP: Tue Jan 26 12:49:04 UTC 2021
 * Starting ntpd: done
 * Starting crond: done
 * Generating ssh host keys: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface wlan0 has IP address www.xxx.yyy.zzz/24
 # Default gateway is www.xxx.yyy.1
 # DNS server address is www.xxx.yyy.1
[root@meye-f7e06d23 ~]#
-----
Image 8GB microSDCard with Win32DiskImager, using 20200606 and my standard
wpa_supplicant.conf file and empty ssh.txt file
boot log:
-----
Using username "admin".
Welcome to meye-f7e06d23!
[root@meye-f7e06d23 ~]# cat /var/log/boot.log
---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Creating data partition: done
 * Formatting data partition: done
 * Mounting data partition: done
 * Copying data skeleton: done
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Setting empty root password: done
 * Copying version file: done
 * Setting UTC timezone: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: none
 * Starting wpa_supplicant: done (brcmfmac)
 * Configuring wireless network: dhcp
 * Setting current date using SNTP: Tue Jan 26 13:04:13 UTC 2021
 * Starting ntpd: done
 * Starting crond: done
 * Generating ssh host keys: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface wlan0 has IP address www.xxx.yyy.zzz/24
 # Default gateway is www.xxx.yyy.1
 # DNS server address is www.xxx.yyy.1
[root@meye-f7e06d23 ~]#
-----
Both images booted normally, using the onboard wifi.
my standard wpa_supplicant.conf file:
-----
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID"
    psk="passphrase"
    key_mgmt=WPA-PSK
}
----

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-01-26:
Same way as a normal image. Since the password is 'blank' you need to do a
clear cache before going to the WebGUI if you changed the password and told
the login to remember you.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-01-26:
So when I use the dev version, using the standard supplicant, as usual, no problems on boot... Wifi connects, IP is assigned, just like normal.  Again, my Windows PC nor my Phone can connect to web interface.  However my media server RPi 4 can.  When I open up the web interface, I have to logout and log back in to access any of the settings (before hand all the settings were greyed out and it had a password set on the admin account that I didn't set so I couldn't access any settings).  To note... at this point, just on wifi, I can ping internet sites, and I can ping other devices on my network, just not my Windows PC and Phone.  So to me, that rules out there being an issue with the onboard wifi... it works.  

Being able to access it on my phone is the main issue here.  If I can do all the setups via my RPi 4 over the web interface and just use my phone to view as I was doing before that'd be great.  I mean, heck, if I port forward in my router and turn the wifi off on my phone I can view the cams no problem... frustrating.  Really not sure whats going on with it to where internally over my home wifi there seems a block...

starbasessd, 2021-01-26:
I don't know. I can talk to the Pi3B+ from a Win10 laptop, MAC OSX, several
linux boxes, and a tablet and phone, and ping websites in the world, on
various connection types. The Pi3B+ can ssh to several of the devices, too.
My network isn't "quite" standard, though, my camera system is on it's own
subnet.
I just put it on my "normal" network, and it's talking to everything there,
too. If you are using anything like a mesh wifi network, I understand they
have firewalls on each of the nodes that need to be dealt with.  If
traceroute /tracert can't get from point a to point b, that is a router
setup issue (DHCP giving a wrong default gateway the most common) Router
issues can be the modem router or wifi router if different devices...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-03-05:
Nope, no update.  I've been working offshore, but been home the last week and made no headway.  Both RPi 3 B+ devices still are not playing ball and I cannot access the from my windows laptop or my iPhone via the web interface.

starbasessd, 2021-03-05:
<=- to internet --- modem/router --- switch --- WiFi router --- Pis
                                                                  L --- PC
or another arrangement?

BennyIsYourBestFriend, 2021-03-06:
Hi,
you might have found a solution to the WiFi problem by now - I had the exact same issue with my Raspberry Pi 4B and couldn't find help on this site, and I got pretty frustrated of such an annoying problem, so I just tested out some tweaks in the wpa_supplicant.conf -file as I was pretty sure my other setups were spot on. It took some time and beers but the following configuration worked for me:

ctrl_interface=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=Country

network={
    ssid="*********"
    psk="********"
key_mgmt=WPA-PSK
}

Osmii, 2021-03-07:
Hello!
I also had problems with wifi droping(totaly first and new setup,Rpi2 with Tp-link dongle) with latest and other releases. Then I connected to a different AP  on other location and now is working ok.

sectshun8, 2021-03-11:
> 
> 
> <=- to internet --- modem/router --- switch --- WiFi router --- Pis
> 
> or another arrangement?

Much simpler...

<=- to internet --- modem/router ---  Pis

starbasessd, 2021-03-11:
Your modem/router does WiFi too?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-03-11:
> Hi,
> you might have found a solution to the WiFi problem by now - I had the exact same issue with my Raspberry Pi 4B and couldn't find help on this site, and I got pretty frustrated of such an [...]

Appreciate the effort, but you'll see by reading the thread that I've tried that very supplicant.conf already... though not necessarily with the SSID and PSK on the same line... or is that a typo?

sectshun8, 2021-03-11:
> 
> Your modem/router does WiFi too?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Yes.  Coax line comes into the back, it then has 4 ports for wired connections and also broadcasts via Wifi.

BennyIsYourBestFriend, 2021-03-11:
> Appreciate the effort, but you'll see by reading the thread that I've tried that very supplicant.conf already... though not necessarily with the SSID and PSK on the same line... or is that a typo?

Yes, it's a typo... sorry for that.... 
I hope you'll manage to get it working soon again.

starbasessd, 2021-03-11:
I don't know. To me, it's pointing to a routing issue or a cache issue. On
my Windows machine I do route print, and my #1 is
0.0.0.0 [everything not otherwise listed] 0.0.0.0 [all subnets]
192.168.xxx.yyy [my primary router] 192.168.xxx.zzz [my local ip address]
25
for my local network:
192.168.xxx.0 255.255.255.0 192.168.xxx.yyy 192.168.xxx.zzz 281
on my Pi
***@***.***:~ $ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use
Iface
default         router2         0.0.0.0         UG    202    0        0 eth0
default         router2         0.0.0.0         UG    303    0        0
wlan0
192.168.2.0     0.0.0.0         255.255.255.0   U     202    0        0 eth0
192.168.2.0     0.0.0.0         255.255.255.0   U     303    0        0
wlan0
both can talk to each other without one having to ping the other...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sectshun8, 2021-03-11:
I'll delve more into it in about a month when I'm back home and can mess around a bit.  Currently headed back offshore so can't really do much.

sectshun8, 2021-04-29:
So... back at this.  I had a completely separate IPC360 camera in my daughters room I could never get on the wifi before... and I gave that one another go recently, and it just joined no problem.  So I thought, hmm... I'll give my two RPi's a go again with the newest iteration of mEOS and see what happens.

Believe it or not, both my RPi cameras are working again with no real explanation why.  I can say instead of using the supplicant to join the Wifi straight away, I set them up on my wired network first, then enabled the Wifi setting in the mEOS via the web interface and entered in the appropriate credentials.  That would really be the only thing I did different... but that being said, I tried this months ago with no luck, so who knows.

Anyhow, both RPi 3B+'s using image 202006, back up and running.  I really wish we could have come up with a reason why to help anyone in the future, but that doesnt look like it's going to be the case.  Unless anyone has anything add this thread can be closed.

closed 2021-04-29 by starbasessd