motioneye-project/motioneyeos #2648: Low fps with RTSP streaming

Poli88, 2021-01-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026

### Board Model

I am using the following board/model: RaspberryPi 4

### Camera

I am using the following type of camera: Network Camera.

My camera model is: Supereye PC200

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Log Files

I consider the following log files relevant to this issue:
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5843738/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5843739/motion.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5843740/messages.log)

Hello, I recently made a fresh install of MotioneyeOS on my RaspberryPi4 and added my 4 x IP Cameras using the RTSP address. Looking at the various posts, I have a quite common issue with this system: I experience low fps compared to the actual stream. 
Could you please help me ???

TheCalmingClam, 2021-01-29:
This is my first post on this forum, and github more broadly, so I apologize if it isn't proper etiquette to add on to a thread like this. However, I'd also like to add my experience to this post. 

I'm using an rpi4 with motioneyeOS version 20200606 and an ethernet connection that connects back to my router. I get ~30 fps when watching the live stream of my IP camera in the motionEye web interface, but when it detects motion and starts to record, the resulting .mp4 video files report that their frame rate is ~2 fps (as seen in the properties > details tab in Windows). I have tried using each video formats available as an option in the settings, and none of them get above ~2 fps when recording. I would prefer not to put it in pass through mode, as my understanding is that this will remove the timestamp watermark from the video. My question is; is this a limitation with the rpi4 hardware, or are there settings changes that I'm not aware of that will solve this issue? Or is there a fix for this issue coming in the next stable release of motionEyeOS? 

Thanks for all your time and effort into this project!

starbasessd, 2021-01-29:
mp4 is highly compressed, which takes a lot of cpu/gpu horsepower to do so.
You can try a different video format to save in and not lose the timestamp.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheCalmingClam, 2021-01-29:
Thanks for your feedback. I may be misunderstanding, but I had mentioned in my previous post that I have tried all the other video formats available and they all result in videos that have about 2 fps.

jasaw, 2021-01-31:
@TheCalmingClam You have 2 options:
1. Use OMX version of the encoder (mp4/OMX). This will offload the video encoding to the GPU. It may not be able to do 30 fps, but should give you much higher frame rate. If you have multiple cameras all recording at the same time, the frame rate will drop.
2. Use pass-through recording and lose timestamp in the video.

TheCalmingClam, 2021-01-31:
@jasaw Unfortunately, it's already on the mp4 OMX option, and still getting 2 FPS. I only have 1 wireless IP cam attached to motionEyeOS. I was under the impression that the rpi4 (4gb) could record @ 30fps with motioneyeOS; is that not the case?
[image]
[image]

jasaw, 2021-02-01:
@TheCalmingClam that is really odd. Can you please attach your log files here?

TheCalmingClam, 2021-02-01:
[boot.log](https://github.com/ccrisan/motioneyeos/files/5901741/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5901742/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5901743/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/5901744/motion.log)

Thanks in advance for taking a look.

jasaw, 2021-02-01:
@TheCalmingClam I don't see any trace of your rpi4 recording. All I can see in your log files are:
>netcam_init_jpeg: no new pic, no signal rcvd

Can you please try pass-through recording and see what the frame rate is?

TheCalmingClam, 2021-02-01:
@jasaw Does something change when you put the camera into passthrough mode? I am not able to alarm it to start recording, even though it still says it's in "motion triggered" mode. 

[image]

jasaw, 2021-02-01:
@TheCalmingClam pass-through recording should not affect motion detection. It might be something to do with auto threshold tuning? I never use auto threshold tuning before, so maybe someone else can chime in here.

TheCalmingClam, 2021-02-02:
I turned off auto threshold and got it to record on motion; however, still 2 frames:

[image]
[image]

jasaw, 2021-02-02:
@TheCalmingClam I think your rpi4 is only receiving 2 fps from your IP camera. How is the rpi4 getting the camera feed? What protocol is the feed, RTSP or MJPEG or RTMP? Some IP cameras have multiple "end points" where one end point implements high res high frame rate for recording, and another end point for low res low frame rate for motion detection. Check that you have the correct end point.

TheCalmingClam, 2021-02-02:
@jasaw That's certainly possible. I am not exactly sure how to tell which protocol it's using. Here's the URL from the "Camera Device" field in the motioneye settings, if that helps: http://192.168.1.253:88/cgi-bin/CGIProxy.fcgi?cmd=snapPicture2&usr=Username&pwd=Password - it's a foscam fi9800p wireless IP cam.

Also, I just wanted to note that the streaming FPS rate is usually in the 20s, as seen here:
[image]

starbasessd, 2021-02-02:
Try:
rtsp://[ip_address]/videoMain
or
rtsp://[ip_address]:88/videoMain
username & password in the usual boxes
I have the 9900 and the second works for me.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-02-02:
https://www.ispyconnect.com/man.aspx?n=Foscam

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

TheCalmingClam, 2021-02-02:
I setup a new camera with one of the RTSP URL's provided and with otherwise identical settings to my previous camera, and the first 2 recorded videos report a frame rate of 25fps and then 9fps. This is a huge improvement over 2fps! I really appreciate everyone's help, thanks so much! And a special thanks to the developers for making a product that is super easy to setup!

TheCalmingClam, 2021-02-02:
Sorry to hijack @Poli88's thread, but hopefully his issue is the same one I had. For anyone in the future searching this, I have a Foscam FI9800P wireless IP camera connected to motionEyeOS and the setting that improved my frame rate was using a URL of rtsp://192.168.1.x/videoMain and then putting username and password in the respective fields. 

Thanks again!

closed 2021-02-10 by starbasessd