motioneye-project/motioneyeos #2649: writeimage.sh error mounting boot partition

rullywowpcb, 2021-01-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
Yes. I have read.
### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi-20200606.img

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: Rpi Camera

My camera model is: Rpi Camera

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
n/a
### Log Files

OS: 
I am using Pop!_OS 20.04

I consider the following log files relevant to this issue:

n/a

I am having difficulty writing the image to the SD card with writeimage.sh. It seems to write the image ok but gives an error when trying to mount the card. 

I am using the command: 
`./writeimage.sh -d /dev/sdd -i "motioneyeos-raspberrypi-20200606.img" -n 'myssid:mypassword'
`
Results in:
` * writing disk image to sdcard
330+0 records in
330+0 records out
346030080 bytes (346 MB, 330 MiB) copied, 10.0539 s, 34.4 MB/s
 * re-reading sdcard partition table
 * mounting sdcard
mount: /home/rullywow/Downloads/motioneyeos/.boot: wrong fs type, bad option, bad superblock on /dev/loop0, missing codepage or helper program, or other error.
`
I have tried this command with sudo as well as being root and the same result. The card seems to flash but the boot partition never gets mounted properly, so I cannot add a wpa_supplicant.conf as we normally do with RaspberryPi.

Here is the result of a fdisk -l
`Disk /dev/sdd: 29.83 GiB, 32010928128 bytes, 62521344 sectors
Disk model: SDDR-B531       
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x9e3f5cfb

Device     Boot   Start      End  Sectors  Size Id Type
/dev/sdd1  *       2048    63487    61440   30M  e W95 FAT16 (LBA)
/dev/sdd2        204800   655359   450560  220M 83 Linux
/dev/sdd3       2097152 62521343 60424192 28.8G 83 Linux
`

Any help would be greatly appreciated, thanks in advance!

starbasessd, 2021-01-21:
If you use the wifi option in writeimage.sh, you don't need to put the
wpa_supplicant.conf file in the SDCard /boot partition
If you create a tmp folder:
sudo mkdir /meostemp [enter]
then mount the /boot partition there:
sudo mount /dev/sdd1 /meostemp [enter]
then cp the wpa_supplicant.conf to that folder:
sudo cp /[path_to_wpa_supplicant.conf]/wpa_supplicant.conf /meos/temp
[enter]
[to enable ssh access do:   touch /meostem/ssh [enter] ]
then unmount the sdcard
sudo umount /meostemp
should give you the results you are looking for.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-04:
No further updates after > 30 days

closed 2021-03-04 by starbasessd