motioneye-project/motioneyeos #2650: External Hard Drive (WD Black P10) is not Detected.

napra77, 2021-01-21:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (motionEyeOS 20200606).

### Board Model

I am using the following board/model: (Raspberry PI 4).

### Camera

I am using the following type of camera: (Network Camera).

My camera model is: (Eufy Security 2K Indoor Cam).

### Network Connection

My motionEyeOS unit is connected to the network via: (Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
(External Hard Disk: WD Black P10 2TB Drive)

### I'm not seeing the hard drive in the drop-down menu. I tried a Sandisk USB drive and it showed up fine. 

### Log Files

I consider the following log files relevant to this issue:
[motion.log](https://github.com/ccrisan/motioneyeos/files/5845964/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5845965/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5845966/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5845967/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5845968/dmesg.log)

starbasessd, 2021-01-21:
It is possible that the drive pulls too much power for the Pi  or power
supply can provide to spin it up and initialize. Please check it with
RaspberryPiOS.
What size, format and type Sandisk is recognized?
Have you tried it with an externally powered USB Hub?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

napra77, 2021-01-21:
> It is possible that the drive pulls too much power for the Pi or power supply can provide to spin it up and initialize. Please check it with RaspberryPiOS. What size, format and type Sandisk is recognized? Have you tried it with an externally powered USB Hub?

Power could be an issue theoretically I guess, but the Pi has a 3A power supply, and this is a USB 3.0 portable drive so I didn't think it would be a problem.

I used a USB 3.0 SanDisk 256GB drive I had.

Unfortunately I don't have a powered USB hub that I could try. 

Also in the dmesg.log, I see some lines that indicates to me at least, the drive is detected fine:

[    1.102094] usb 2-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[    1.133219] usb 2-2: New USB device found, idVendor=1058, idProduct=2630, bcdDevice=40.00
[    1.133234] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.133246] usb 2-2: Product: Game Drive
[    1.133258] usb 2-2: Manufacturer: Western Digital
[    1.133270] usb 2-2: SerialNumber: 575839324136304E30534834
[    1.136728] usb-storage 2-2:1.0: USB Mass Storage device detected

starbasessd, 2021-01-21:
Depending on your power source, it may or may not be putting out 15w or 3
amps. Especially if you are using a 'charger' and not a 'power supply'.
USB-C implementation on Pis does not include circuitry to let chargers know
it needs more power and they tend to default to 2amps.
https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md

Spin up timing can be an issue. If you can mount the drive manually after
the full boot:
sudo mkdir /data/test [enter]
sudo mount /dev/sda1 /data/test [enter] [assuming the drive is partitioned
as ext4 or fat32 or exfat]
df -h [enter] [shows if drive mounted]
If this was successful, then the issue is it's not discovered during power
up.
There is a change you can make in the eeprom config to add time to allow
spin up, but it requires you to make it in RaspberryPiOS, as motionEyeOS
doesn't have the editor capability.
After the eeprom is changed, then you can run motionEyeOS normally.
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md
USB_MSD_DISCOVER_TIMEOUT

If no USB mass storage devices are found within this timeout then USB-MSD
is stopped and the next boot mode is selected

Default: 20000 (20 seconds)

THere are other settings, too. Make sure the timeout for USB Power Off is
set long enough to reboot the USB WD Drive.

I did a quick google for RP4 and WD drives and there are a lot of them out
there.

I have similar issues in other areas with non SSD drives and non USB Stick
thumb drives. Make sure the drive is NOT partitioned with GPT.

Any multi-TB drives I partition as multiple 1TB partitions. Seems to work
for me. in general. I use NAS storage instead of local for my motionEye
hubs.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

napra77, 2021-01-22:
I figured it out. It's a problem with the file system. I  got a clue when I tried a different USB drive that didn't get recognized. The HD and this USB drive were both formatted in exFAT which MotionEye doesn't recognize. Once I reformatted them to either FAT32 or NTFS, both drives got picked up fine.

closed 2021-02-03 by napra77