motioneye-project/motioneyeos #2651: Switching to a second network adapter, possible?

jl2008, 2021-01-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * Alfa Long-Range Dual-Band MIMO Wireless USB 2.0 Wi-Fi Adapter, https://www.amazon.ca/dp/B00V5KSIZS/

I'd like to try using the external adapter (listed above), if possible, instead of the built-in WiFi as sometimes the signal is too weak and while images are recorded, notifications are not sent. Ironically, that happens when there are people between the RPi box and the router, the extra absorption is enough to prevent the notifications from being sent.

Currently, the system uses `wlan0`. 

My questions:

- How can I find out if the adapter is supported? `wlan1` appears in the output of `ifconfig `and `iwconfig`, as below, but does not receive an IP address from the router's DHCP server.

- If supported, how can I switch the configuration to it?

Thank you.

```
[root@meye ~]# iwconfig

wlan1     IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=0 dBm
          Retry short  long limit:2   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off

wlan0     IEEE 802.11  ESSID:"<redacted>"
          Mode:Managed  Frequency:2.427 GHz  Access Point: <redacted>
          Bit Rate=39 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=40/70  Signal level=-70 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:29  Invalid misc:0   Missed beacon:0

[root@meye ~]# ifconfig -a

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.2 netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe28::27ab:228f:111f:77e2  prefixlen 64  scopeid 0x20<link>
        ether <redacted>  txqueuelen 1000  (Ethernet)
        RX packets 11415  bytes 789039 (770.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 46704  bytes 70979929 (67.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether <redacted>  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

starbasessd, 2021-01-23:
Interesting.
I used a Pi3B+, and motionEyeOS 20200606
installed my wpa_supplicant.conf file and ssh.txt files in /boot
No ethernet connection
Booted normally, no secondary WiFi dongle installed.
Pulled wifi ip address X.Y.Z.138 for MAC b8:27:eb:qq:rr:ss
Installed edimax wifi dongle
Confirmed in dmesg that it mounted it correctly
didn't auto pull an IP address.
Rebooted
This is where it gets interesting.
Can't ssh or WebGUI to the X.Y.Z.138 address
Look at DHCP server leases, see a new one for a different MAC but with the same name
ssh to X.Y.Z.113, and lo and behold, it connects
do an ifconfig, and it only shows 1 wlan, wlan0, but the mac is now 74:da:38:ll:mm:nn, which is the edimax's mac
and is perfectly operational.
Shut everything down, and switch to another dongle (This time a Xerox test Wifi-N)
Yep, new connection, shows MAC 9c:93:4e:hh:ii:jj and IP X.Y.Z.147
Ditto for ifconfig, only the one wlan0, and the dongle's IP & MAC.
It may be that your dongle is acting the same, or not fully recognized...

jl2008, 2021-01-23:
Hmm.. interesting indeed. I have an Edimax as well and I saw the same behaviour: upon reboot, the Edimax becomes wlan0 and the built-in adapter is not activated. The Alfa is seen, but it stays as wlan1 and not active. I'll try with the Edimax for a while, not sure if meaningfully better over the built-in one since it doesn't have a stick antenna.

BTW, is it possible to enable the second one at all? Not asking about it being used by MotionEyeOS, just about the activation.  `ifup `doesn't work:

```
[root@meye ~]# ifup wlan1
ifup: can't open '/etc/network/interfaces': No such file or directory
```

Thanks.

starbasessd, 2021-01-23:
Just ran it again with a stick-type wifi dongle and it had the same
behavior as my other 2 dongles.
For grins and giggles, could you please:
ssh or terminal in
run command
logrotate -f /etc/logrotate.conf [enter]
reboot [enter]
Then in the WebGUI, in Expert Settings attach all 5 logs to this issue.
Please?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jl2008, 2021-01-23:
Done. How can I resist such a request? :)

[motion.log](https://github.com/ccrisan/motioneyeos/files/5860424/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5860425/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/5860426/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/5860427/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5860428/messages.log)

starbasessd, 2021-01-23:
Was this with or without the stick dongle?
Does it work with RaspberryPiOS?
Everything I'm finding shows it's abandoned Hardware (last documented
around 2015) The linux drivers and firmware are virtually non-existant and
very inconsistent. The RT5732 requires firmware for Linux, but I can't find
a binary blob download like the ZD1211rw sticks I have (from 2010 or so) If
I install in RaspberryPiOS, I can grab a firmware folder and pass it to
motionEyeOS and get it to work
If you can get it installed in RPiOS and get it working, we can try this
method...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jl2008, 2021-01-23:
Sorry, forgot to mention: this is with the Edimax adapter in, which works. Yeah, it's older, I had it in a box. I initially wanted to get it working with the Alfa adapter (which has two stick antennas) but know what, it's not worth the effort, I can get a current and supported external WiFi adapter, they are not expensive.

One thing of note is what prompted all this: I suspect that a lack of connectivity at the time of the motion alert causes it not to be sent at all. I want to emphasize that this is just a hypothesis at this time, I haven't confirmed or infirmed it yet. If confirmed, that's more valuable to be fixed. I'll do some tests on my own as well.

Thanks.

starbasessd, 2021-01-23:
Do you have "Attach Pictures Time Span" set to a non-zero number?
This, in combination with the number of pictures it might have to search
through to find the ones to attach, can cause issues. There are a couple of
settings that can be modded to increase the amount of time allowed to
search and attach...

/data/etc/motioneye.conf
you can add some of these: (all numbers are the normal defaults)
list_media_timeout

Configures the timeout in seconds to wait for media files list. Defaults to
120.
list_media_timeout_email

Configures the timeout in seconds to wait for media files list, when
sending emails. Defaults to 10.
zip_timeout

Configures the timeout in seconds to wait for zip file creation. Defaults
to 300.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jl2008, 2021-01-23:
Indeed, I have the "Attach Pictures Time Span" set to 3 seconds. I receive 3 pictures attached (~500KB each).

Other possibly relevant settings: "Motion Gap" is 30s, "Captured Before" 1 frame, "Captured After" 1 frame, "Minimum Motion Frames" 5 frames. Timeouts are as below:

```
[root@meye etc]# grep timeout *
motioneye.conf:remote_request_timeout 10
motioneye.conf:mjpg_client_timeout 10
motioneye.conf:mjpg_client_idle_timeout 10
motioneye.conf:smtp_timeout 60
motioneye.conf:zip_timeout 500

```

starbasessd, 2021-03-04:
Any further updates? If not, please close.

jl2008, 2021-03-05:
I'll close it, thanks.

closed 2021-03-05 by jl2008