motioneye-project/motioneyeos #2652: How can I make wifi camera with Raspberry Pi Pico?

ivanjf1978, 2021-01-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
Hi, 

I just want to know if someone have made a project with Raspberry Pi Pico using it like wifi camera and it could be add on MotionEyeOS

Thank for any advice and help

Regards

Ivan

starbasessd, 2021-01-22:
A] It was just released yesterday.
B] It's a microcontroller, not a SBC like the Pis.
C] None of the accessories are available yet, and it will take at a
minimum, a wifi adapter (none on board) and a camera (also not on board. It
will take an external power supply if you want to use the microUSB for a
USB camera.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ivanjf1978, 2021-01-22:
Thanks. I just want to know if the Raspberry pi Pico it is like the ESP32 that it can be use like wifi camera. And I want to know it is possible to add on motioneyeOS like another camera?

ivanjf1978, 2021-01-22:
I understand ESP32 it is a microcontroller and it can be use like wifi camera. I have used the ESP32 like a wifi camera but I can not add on motioneyOS, that's why I ask if the new RP pico can be use like the ESP32?

have a look

https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/

starbasessd, 2021-01-22:
The only people today who have that info are:
The Raspberry Pi Foundation
Companies like AdaFruit, PiMoroni, and the other companies who will be building the Pico. 
Like I said, it was just released YESTERDAY (21 Jan 2021)  to the General Public. I've seen a couple of YouTubers who got pre-release versions, but even they don't know the full capabilities.

starbasessd, 2021-01-22:
If the microcontroller can output a stream or jpeg in a format that
motion/motionEye can understand, then it is possible. Most webcams do not
have a microprocessor, just a microcontroller, and someone took the time to
write the software needed to do the correct output support.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-01-22 by starbasessd

langyxxl, 2022-09-19:
Maybe pico w and arducam can work together to provide such feature.

starbasessd, 2022-09-19:
Pico is not compatible with motionEyeOS. It is not a full computer. It does not have the capabilities needed, such as CSI, GPU, etc...
There are a number of cameras available but they are unable to run software such as motionEye: [See Here](https://www.arducam.com/raspberry-pi-pico-camera-modules/)