motioneye-project/motioneyeos #2653: Failed to upload File / GDrive

YuenTW, 2021-01-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (motionEyeOS dev20201026).

### Board Model

I am using the following board/model: (Raspberry Pi Zero W).

### Camera

I am using the following type of camera: (NMAL Camera).

My camera model is: (Raspberry Pi Camera V1).

### Network Connection

My motionEyeOS unit is connected to the network via: (WiFi).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

 * [motioneye.log](https://github.com/ccrisan/motioneyeos/files/5858145/motioneye.log)

 * [boot.log](https://github.com/ccrisan/motioneyeos/files/5858157/boot.log)

Question here : 
So ive been using MotionEye OS for 4 - 5 days now, i saw the problem that MotionEye could auto upload 30 - 45 min, but yesterday i set the movie max second to 3600sec (1hr), when i check the MotionEye.log, it will show up this : 

ERROR: failed to upload file "/data/output/Camera1/2021-01-23/02-00-10.mp4" with service gdrive: 
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 97, in upload_file
 
Im not sure if its the system problem or its the problem with wifi/gdrive but i need some help here, im so confused

starbasessd, 2021-01-22:
At the console (keyboard & mouse) or a SSH session, run the command: df -h
How large of a microSDCard are you using?
Does it create the local copy of the video file?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

YuenTW, 2021-01-22:
I have an SanDisk 64GB of microSDCard in the Pi rightnow
And yes it do create a local copy of the video file, and i could download it.
Sorry im not quite sure why.. but i cant login to my ssh. I have type the correct password into the password, but it still wont let me in (Permission denied (publickey,password,keyboard-interactive)

starbasessd, 2021-01-22:
Can you open the most recent file?
THere is no password unless you added an admin password at some point.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

YuenTW, 2021-01-22:
Yes i could open, if im not wrong about your talking about.
i do reset my password to none on admin password, but it somehow still not work for me.
[image]

starbasessd, 2021-01-22:
username is admin or root

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

YuenTW, 2021-01-22:
Still nothing worked. :(
do i need to get like "ssh_key" file to my mac so i could login? or i do not need this and i could just login via ssh with admin/root?

starbasessd, 2021-01-22:
No. On my MAC in Terminal, I use:
ssh admin@[ipaddress] [enter]
It prompts me for password for admin, I hit [enter]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

YuenTW, 2021-01-23:
Sorry for the late reply, I'm at GMT+8 time zone.

I've got in to the ssh now, this is the info that I got it

[image]

YuenTW, 2021-01-23:
Update : in the picture you could see the file that auto recored every 1hr
[image]

But atm in my GDrive,
[image]

I just have 2 of the hour file that have uploaded to GDrive, And its been like 5-8 hr right now, other file didn't auto upload to the Drive, The new MotionEye Log showed up saying : File is to large. anyway to fix this?

[motioneye-2.log](https://github.com/ccrisan/motioneyeos/files/5859734/motioneye-16.log)

starbasessd, 2021-01-23:
Looks like GDrive isn't allowing you to upload files larger than 1GB
Try reducing the recording time to 20 minutes to see if that helps.
If I get a chance today or tomorrow, I'll build a Pi with your specs and
see what I can get.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

YuenTW, 2021-01-23:
Got it, i'll change the recording time to 20 minutes.
ill keep update if its working or not.

YuenTW, 2021-01-23:
Hey, so the 20 minute video have been uploaded to GDrive and its all worked nice, but it somehow just upload 2 of the file into GDrive.
 
[image]
[image]

Not sure if it is still uploading or else, but everything works fine! 
I would like to still let the time go up to 1hr per video, thats good for me if i need to check what time what thing happend in the specified time.

Here is the new Log file to let you check. [motioneye-18 (1).log](https://github.com/ccrisan/motioneyeos/files/5860656/motioneye-18.1.log)

starbasessd, 2021-01-23:
THere is an issue about sending motion notifications, too, and this, in
combination with the number of pictures it might have to search through to
find the ones to attach, can cause issues. There are a couple of settings
that can be modded to increase the amount of time allowed to search and
attach...

nano /data/etc/motioneye.conf
you can add some of these: (all numbers are the normal defaults)
list_media_timeout

Configures the timeout in seconds to wait for media files list. Defaults to
120.
list_media_timeout_email

Configures the timeout in seconds to wait for media files list, when
sending emails. Defaults to 10.
zip_timeout

Configures the timeout in seconds to wait for zip file creation. Defaults
to 300.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

YuenTW, 2021-01-23:
umm so i tried to put all the commend line into the conf. and it just broke... sorry but im like srsly dumb..
do i need to reinstall the MotioneyeOS again? i could still connect to ssh tho... but config got cleared.. 
[image]
Big Yikes..

starbasessd, 2021-01-23:
Shouldn't have 'broke', you mis-typed the filename path so there shouldn't
be an issue...
re-type it correctly...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

YuenTW, 2021-01-23:
[image]
Yeah im sorry i typed wrong xD but i retype it and still not anything showed up and it shown "New File"

starbasessd, 2021-01-23:
re-image.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

YuenTW, 2021-01-23:
sorry, wdym

YuenTW, 2021-01-23:
Sorry for asking again, Do i need to add second after the command? Ex: list_media_timeout_email 120

YuenTW, 2021-01-23:
I have set all you recommend to the line now.
[image]
and also i have changed the time record to 3600 second

YuenTW, 2021-01-24:
So the file per hour is reduce to 600 - 800 MB per movie now, but some how it still get failed to upload to GDrive

[motioneye_3.log](https://github.com/ccrisan/motioneyeos/files/5862530/motioneye_3.log)

starbasessd, 2021-01-24:
File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line
97, in upload_file
MemoryError

Try rebooting. If that fails, reimage the SDCard

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-04:
No updates in > 30 days

closed 2021-03-04 by starbasessd