motioneye-project/motioneyeos #2656: SSH Bug after restore a config

jariojose, 2021-01-26:
Hello, all!

I am using motionEyeOS 20200606 in a raspberry pi 3b+. 

I think that there is a bug when restore configs. When I am using a fresh install, without cameras or personal configs, the ssh connects well. But If I restore a config, using my cameras, personal password, etc. the ssh fail to connect.

Messages Log show: 

> Jan 26 09:31:50 meye-f7d6fde3 auth.info sshd[1481]: Failed password for admin from 192.168.1.144 port 54466 ssh2
> Jan 26 09:31:51 meye-f7d6fde3 auth.info sshd[1481]: Connection closed by authenticating user admin 192.168.1.144 port 54466 [preauth]
> Jan 26 09:31:55 meye-f7d6fde3 auth.err sshd[1623]: error: Could not get shadow information for admin

 
[motion.log](https://github.com/ccrisan/motioneyeos/files/5873314/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/5873316/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/5873309/messages.log)

starbasessd, 2021-01-26:
Was there an admin password set in the system you are restoring from?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jariojose, 2021-01-26:
Yes. There is a encrypted password. 

motion.conf inside the tar: 

> # @admin_username admin
> # @admin_password <HERE_Encrypted_password>

starbasessd, 2021-01-26:
> Yes. There is an encrypted password.

Does the unencrypted password work in the WebGUI?
It is possible, that due to the change in the system SSH key that it can't
unencrypt it properly, not sure. If it works in the WebGUI, re-set it
there, and see if that fixes it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jariojose, 2021-01-26:
Unsuccessfully.

The WebGui works fine. I changed encrypted to unecrypted but didn't work.
I used too encrypted to login in ssh, but doesn't work.

starbasessd, 2021-01-26:
Can you try to ssh in with admin, and just hit [enter] for the password?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jariojose, 2021-01-26:
Unsuccessfully too. 😂

I send the files before restored. So, I restored. I can't to access ssh, only webgui.

starbasessd, 2021-01-26:
I'll test in a few.
If you install dev20201026, can you use the restore files successfully?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jariojose, 2021-01-26:
Thank you, @starbasessd, but unsuccessfuly again.

Yes, I can restore. I did it but I can't access ssh yet, when last config are restored. :/

starbasessd, 2021-01-26:
Created new Rpi3B+, 20200606 image, 3 network cameras
Changed admin password
Confirmed able to ssh in.
Ran backup (Settings, General Settings, Backup)
re-image with 20200606
confirm access with no password in both ssh & WebGUI
restore just created image with 3 cameras
ssh works with password set for backup
WebGUI works with password set for backup.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jariojose, 2021-01-26:
So I'm going to set it up again on a clean install. Thank you!

closed 2021-01-26 by jariojose