motioneye-project/motioneyeos #2657: Vertical Pink Bands in Live View / Blue 'negative' images in recorded motion files

MyopicMonster, 2021-01-26:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
CONFIRMED

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
CONFIRMED

I am running motionEyeOS version: 20200606
I am using the following board/model: R-Pi 4B 4G
I am using the following type of camera: MMAL 
My camera model is: R-Pi V2 standard (NOT noir)
My motionEyeOS unit is connected to the network via: WiFi
Peripherals: USB storage device
### Log Files
[bootJan26.log](https://github.com/ccrisan/motioneyeos/files/5876084/bootJan26.log)
[motionJan26.log](https://github.com/ccrisan/motioneyeos/files/5876085/motionJan26.log)

Hello, Im having a persistent problem with both the Display Image and the Recorded (captured) Image.
When viewing the live view, there is frequently vertical pink bars appearing in the view. Size, quantity, location on screen and frequency of the pink bars changes - no pattern but it does sometimes trigger a motion event capture. Sometimes the pink artifacts appear every few seconds on the screen, sometimes every few minutes. sample of image below.
Additionally, when a motion event creates a captured image, it is blue and black like looking at old-school 35mm negatives.
Manual capture = perfect image.
I have tried the pre-released file and it locked up the Pi device completely. the Pi was effectively bricked. It required a complete new set-up and re-write to the last released version 20200606.

I have tried different camera resolutions, motion detection settings, , different (cheap Chinese) cameras and different 3A power supplies + cables without any change in behavior. Cameras tried have always been connected by ribbon cable-no USB cameras.
settings are below. advice greatly appreciated. Failing resolution, repurposing suggestions for the R-pi appreciated

;)

Device: R-Pi 4b 4G
motioneye: 0.42.1
Motion Version: 4.2.2+gitUNKNOWN
OS Version: motioneyeOS 20200606
camera: R-Pi V2 (NOT noir)
camera Device: vc.ril.camera
camera type: MMAL
resolution: 1280x1024
Frame rate Dimmer: 18
Resolution Dimmer: 85
file storage to attached USB
video streaming: OFF
Still Images-
Quality: 80%
Capture: motion triggered
Motion detection-
Frame Change: 5%
Max Change Threshold: 0
Auto Threshold: off
Auto Noise: off
Noise Level: 15%
Light Switch: 75%
Despeckle: on
Motion Gap: 20
Capture Before: 2
Capture After: 2
Minimum Motion Frames: 10
Mask: off
.
Motion Notification: off
Schedule: off

[image]
[image]

starbasessd, 2021-01-26:
See my other comments, please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

MyopicMonster, 2021-01-26:
I dont see the vcgencmd  bootloader_version option/file anywhere - sorry. Now that i have connected the R-pi to a wired LAN, the pink bars are not appearing, but there is still some blue image files (about 50% of captured images as opposed to 100%)
I will reload the latest pre-release version and report back if it works this time.
hooking up a monitor from the HDMI port is only working when using Rasbian OS, not MotioneyeOS (unless im really missing something).

starbasessd, 2021-01-26:
You should get a command line setup (console) on the Pi with a monitor
attached. With that (or an ssh session) to the pi, then you can log in
using 'admin' and either [enter] for the password, or whatever you set it
to.
Then at prompt, you can enter the vcgencmd bootloader_version command in
motionEyeOS or in RaspberryPiOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?