motioneye-project/motioneyeos #2658: FastNetworkMode with framerate=1 results in black images

mjj4791, 2021-01-26:
### Issue
I am running MotionEyeOS on raspberry Pi ZeroW with CSI connected raspi NoIR camera.
Camera is running with:
* fast network camera=ON (no motion)
* resolution: 640x480 or 1280/720
* framerate=2
* exposure=auto
* AWB=Greyworld or Auto 
* Streaming Protocol: mjpeg

With this setup; the camera is producing an image as expected.
When I switch framerate to 1 (all other settings unchanged); the camera is no longer producing a real image, but a a fully black image.....  
The issue also occurs when having
* 640x480 or 1280x720 (maybe more resolutions)
* AWB=auto or AWB=greyworld

Clients are connected; as visible in streameye.log (but receive the fully black image):
```
2021-01-26 23:42:31: INFO : x.y.z.6:34328: connection closed
2021-01-26 23:42:36: INFO : new client connection from x.y.z.6:34338
```

Switching framerate back to >1 (eg 2) immediately restores the actual camera image.

### motionEyeOS Version

I am running motionEyeOS version: 
* motionEye Version | 0.42.1
* Motion Version | 4.2.2+gitUNKNOWN
* OS Version | motionEyeOS 20200606

### Board Model

I am using the following board/model: Raspberry PI ZeroW

### Camera

I am using the following type of camera:  MMAL/Fast Network Camera

My camera model is: raspi mmal /NoIR camera

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

### Log Files

I consider the following log files relevant to this issue:

 * none; 
 no issues/errors/warnings/info messages found in any of hte log files, which show a difference between framerate=1/n0-image and framerate>1/Image

mjj4791, 2021-01-26:
continuing investigation; could be the pi is rebooting/being rebooted....

closed 2021-01-26 by mjj4791

starbasessd, 2021-01-26:
Out of curiosity, what was the goal of running a single frame per second?

mjj4791, 2021-01-27:
The cam is watching the inside of a birdhouse; so not a lot of movement (yet)
and next to that, it currently has horrible wifi coverage... so I am trying to limit bandwidth for now, while still having an image