motioneye-project/motioneyeos #2659: unable to open video device  zerokam OV5647 5MPx

CrYpT0n1T3, 2021-01-30:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 

0.42.1
--
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

### Board Model

I am using the following board/model: pi zero 1,3

### Camera

I am using the following type of camera:V4L2 Camera

My camera model is: bcm2835 zerokam OV5647 5MPx 120

### Network Connection

My motionEyeOS unit is connected to the network via: wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

netis USB WiFi adapter)

### Log Files

I consider the following log files relevant to this issue:

motioneye.log)
`[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2+gitUNKNOWN Started
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)`

starbasessd, 2021-01-30:
The camera should be auto-discovered and set up as an MMAL camera.
The PiZero, if it is not PiZeroW V1.1 then it will not work with motionEyeOS, as it has no networking capabilities, without making some additions (os.conf) in the /boot (30MB) partition).
Please post the boot.log ( under Settings, Expert )
Since the pic I was able to find shows the ribbon cable permanently affixed to the camera board, please make sure the cable is installed in the correct direction on the PiZero
You can also confirm the camera is working if you hook up a monitor and keyboard, or SSH into the PIZero, and run:
raspistill -t 2000 -o image.jpg
you should then see a file in the current directory
ls -la
If you get an error, post it here.

CrYpT0n1T3, 2021-01-30:
I connected the camera the other way round, sorry, it works

starbasessd, 2021-03-04:
resolved

closed 2021-03-04 by starbasessd