motioneye-project/motioneyeos #2661: I can not see ip camera  , it is  connected with flat cable to raspberry pi 4. Add in the conf file the line with " rpi_camera:". The montioneyes in the HA can not able to see the camera or any picture.

massimoleonardi, 2021-01-31:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-01-31:
MotionEye or motionEyeOS?
MotionEye is at:
https://github.com/ccrisan/motioneye/issues
MotionEyeOS doesn't work with HomeAssistant.
Your camera is not an IP Camera if it is connected to the PI with a ribbon cable, it is a MMAL camera.
If it is not working, the most common issue is the cables are not connect to the PI or the Camera correctly.
If you are having issues with the HomeAssistan version of motionEye you need to contact the HomeAssistant community & FRENCK, as they support their version
https://community.home-assistant.io/t/home-assistant-community-add-on-motioneye/71826

closed 2021-01-31 by starbasessd

massimoleonardi, 2021-02-01:
thanks for support.