motioneye-project/motioneyeos #2662: My Raspberry Pi Zero W running MotionEye OS will not stay coneected to my WiFi.

BrianG61UK, 2021-02-01:
My Raspberry Pi Zero W running MotionEye OS will not stay coneected to my WiFi (A TP-Link EAP225 access point).

Is there some secret command I can add to my wpa_supplicant.conf to make the WiFi more compatible?

Perhaps a way to make it stick to 802.11g.

starbasessd, 2021-02-01:
I did not find any settings applicable to PiZeroW wpa_supplicant.conf, as it is already 2.4GHz (b/g/n) and automagically connects to the AP with your SSID and passphrase.

BrianG61UK, 2021-02-01:
It fairly reliably gets an IP address But then either no data ever passes or when it finishes booting (takes a long time) it connects and I can use it for a maximum of about a couple of hours then connectivity is gone again for good.

starbasessd, 2021-02-01:
If you can get into the console (monitor & keyboard) or SSH into it, try
the command:
vcgencmd get_throttled
Also, fplease install RaspberryPiOS Lite, with nothing running (but if you
have anything else connected leave it connected) and see if the same thing
happens. Use it in the same location, and use the same power supply. This
will help diagnose software vs hardware issue.
If you are running motionEyeOS 20200606 please downgrade to 20190911, or
upgrade to dev20201026, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Johnny-quattro, 2021-02-20:
I would also check to see if ipv6 is enabled; I've had issues before with Pi computers dropping off the network due to ipv6 issues.
I don't use ipv6, only ipv4.

Check output from 'ifconfig' and if you want to disable the use of ipv6, add the following to /etc/sysctl.conf:

net.ipv6.conf.all.disable_ipv6=1
net.ipv6.conf.default.disable_ipv6=1
net.ipv6.conf.lo.disable_ipv6=1
net.ipv6.conf.eth0.disable_ipv6=1     #optional i.e. not needed if wireless/wifi only
net.ipv6.conf.wlan0.disable_ipv6=1

You can re-read the sysctl.conf and therefore not have to reboot to make the changes active by issuing the following command:

sudo sysctl -p

starbasessd, 2021-03-04:
Status?

BrianG61UK, 2021-03-05:
Having missed the event I originally wanted this for it's been put aside for now but I will investigate further at some point.

starbasessd, 2021-03-05:
If you are still having issues, please open a new issue.

closed 2021-03-05 by starbasessd