motioneye-project/motioneyeos #2663: No Wifi connection

Mjay86, 2021-02-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I have tried running motionEyeOS version: 20200606, 20190111, 20190427

### Board Model

I am using the following board/model: pi 3b +, pi zero W

### Camera

currently no camera 

### Network Connection

Ethernet is working fine.
WiFi not connecting 

Hi Team,

I currently try to set up the wifi on my pi 3b+ and zero w from the first boot as the zero has no ethernet. 
I have tried it on 3 different versions with the pi 3b+ but I was not success full.

If I use ethernet with the pi 3 it works perfectly fine, I also can set up the wifi in motion reboot, and the wifi works as well.

I have set up 2 separate internet one with a password that has only numbers and one that has only letters but non worked. 

I created the wpa_supplicant as following 

country=AU
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
	scan_ssid=1
	ssid"raspberry"
	psk"raspberry"
}

I have also tried to use capital letters with the .CONF 

Any help would be appreciated
[wpa_supplicant.conf.txt](https://github.com/ccrisan/motioneyeos/files/5916667/wpa_supplicant.conf.txt)
[image]

.

Not sure how i can search for the log if i don't have an IP address and the pi is not connected to the internet.

Thank you.

starbasessd, 2021-02-03:
You need to remove the .txt extension from the file name.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Mjay86, 2021-02-04:
I was creating the file and deleted the .txt but window 10 seems to put it back in. So I created the same file on a mac and now it works for some reason. 

But thank you for your help

closed 2021-02-04 by Mjay86