motioneye-project/motioneyeos #2664: pi zero slow in starting a python script.

kthemall, 2021-02-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606.

### Board Model

pi zero w

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: original pi camera 

### Network Connection

My motionEyeOS unit is connected to the network via: wifi 

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 

hi, i installed successfully motioneyeos and it runs fine but i have one problem, i use it for open and close a air-pressured "cat-remover" *G*

i start a python script that opens a valve for air-pressure but the script does not start when motion is recognized but sometimes 5-10 seconds later. how can i shorten time between recognition and starting the script?

thx tom

starbasessd, 2021-02-03:
Don't record video on the PiZeroW,
Shorten the amount of video.
Pre-compile the python script.
Use a second PiZeroW to monitor for motion and record the stream from the
first (or reverse the roles...)
Use a faster Pi (Pi3B+ or Pi4B)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-04:
If you need further assistance, please open a new issue.

closed 2021-03-04 by starbasessd