motioneye-project/motioneyeos #2665: Hook when movie deleted

belese, 2021-02-03:
Hello, 

motion has hook for on_movie_start and on_movie_end.
I use that to populate a database with recording.

i'm looking for a hook 'on_movie_deleted'.
Looks like, it's not motion, but motioneye that delete older files.

i can scan  in my app if file has been deleted every X, but it add a delay to synchronize DB with recording files.

Are you interested for a hook 'on_file_deleted' in motioneye? 
I could probably develop  it myself and do a push request, but as i would like to use 'regular' motioneye, and if there's no interest in this, i'll do scanning in my app instead.

Kr

Johan

starbasessd, 2021-02-03:
What would be the use case for such a link?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

belese, 2021-02-03:
as i use a database to store recording info, remove recording from database if file has been deleted.

starbasessd, 2021-02-03:
Curious.
How does it handle deletions by other methods? (using rm -rf ...)
Can it handle non-standard file locations?
Could it be used in both motionEye and motionEyeOS?
Could it be handled by a cron job better? (hourly or daily?)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

belese, 2021-02-03:
i don't understand, 
motioneye already remove older files, but external app can't be advertise when it happen.
And yes, i could disable it in motioneye, and delete older records myself. 
But as it's already implemented in motioneye, i was asking if there's interset in this.

starbasessd, 2021-02-03:
I'm just trying to see if I have interest, trying to figure out if it's
worth it to my.
I am not putting it down, just trying to see if it would be interesting to
me.
I'm still working on better method(s) to delete files when drives get
full... Thought this might be helpful.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-04:
If you are going to submit a pull request, leave it open, but there hasn't been any other interest shown...

belese, 2021-03-04:
no, i've done other way. i'll close it. Thanks.

closed 2021-03-04 by belese