motioneye-project/motioneyeos #2666: Recordings movies stop after a short while

napra77, 2021-02-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606

### Board Model

I am using the following board/model: Raspberry PI 4

### Camera

I am using the following type of camera: Network Camera: Eufy Security 2K Indoor Cam

My camera model is: Eufy Security 2K Indoor Cam

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
External Hard Disk: WD Black P10 2TB Drive

### Log Files

I consider the following log files relevant to this issue:
[boot (1).log](https://github.com/ccrisan/motioneyeos/files/5920099/boot.1.log)
[motioneye (2).log](https://github.com/ccrisan/motioneyeos/files/5920100/motioneye.2.log)

 
[image]

### As you can see from the screenshot, the recording stops after 3 movie clips. If I change some settings, it will restart, but again stop after 3 or so clips. I tried a external hard drive and a different USB stick, but got the same behavior.

starbasessd, 2021-02-03:
I would suggest using dev20201026 instead of 20200606, as there are a bunch
of little weirdnesses. I'd suggest going back to 20190911 if you were using
a Pi3B+ or earlier, as support for the Pi4 didn't get stable until newer
images.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

napra77, 2021-02-04:
> I would suggest using dev20201026 instead of 20200606, as there are a bunch of little weirdnesses. I'd suggest going back to 20190911 if you were using a Pi3B+ or earlier, as support for the Pi4 didn't get stable until newer images.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Well, I tried your suggestion and tried the dev image. It allowed me to add the first camera. Then I went to change the recording settings but "Apply" button would not appear. Reboot doesn't work either. Shutdown worked, and when I rebooted, the whole camera settings panel is grayed out and nothing can be changed. Managed to grab the logs, but nothing else works.
[motioneye (3).log](https://github.com/ccrisan/motioneyeos/files/5927719/motioneye.3.log)
[boot (2).log](https://github.com/ccrisan/motioneyeos/files/5927720/boot.2.log)

starbasessd, 2021-02-04:
Make sure you do a clear cache if you previously set a password on
motionEye in a previous version...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

napra77, 2021-02-04:
> Make sure you do a clear cache if you previously set a password on motionEye in a previous version...
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Ok, thanks! That was the issue. Cleared cache and everything seems to be working fine including the recordings.

closed 2021-02-10 by starbasessd